TobKra96/music_led_strip_control #19: Whats the use of "Use Music Client"

skyzonemc, 2020-03-23:
I don't understand what it is for. Would be nice if you told me (:

PS:
Do you have Discord or something were I can easily contact you?

TobKra96, 2020-03-25:
Hi,
with the music client, you can use a pc to calculate the effect.

Music Client (Get audio stream via mic or virtual mic like stereo mix, calculate the audio processing and the led array) --> Server (get the array and send the signal to the output pin)

This feature will be changed/extended in v2. There will be an option to have multiple esp2866 output devices.

PS: Yes I have discord. My name: TobKra #6804

skyzonemc, 2020-03-26:
ohhhh, okay. Got it working on my windows pc. Is it possible that the effects aren't exactly the same in the client?

btw, sent a friend request on discord c;

Bigwheel13, 2020-09-21:
I understand what the client is but this is my first time using a stream for input. Could you point me to a guide or give info on what I need to do? I haven't been able to find anything.

magifan, 2020-10-26:
Hello tobkra, I love your led project, I'm a huge fan. I'd like to offer support for your version 2. I'm not a good programmer yet, but I'm aiming to become (in training rn). Please message me back - I also texted you on discord

TobKra96, 2020-11-06:
Hello magifan,

sorry for the late response.
That's nice to hear, that you want to help me.

I created a discord server.
https://discord.gg/r9Gu6YTU

Have a nice day
TobKra

closed 2021-01-03 by TobKra96