TobKra96/music_led_strip_control #20: trying to run MLSC

ayntot, 2020-04-18:
Hello, 

I seem to be having issues getting the MLSC to run. I get to the  /share/music_led_strip_control/server and enter: sudo python3 main.py. Nothing happens I don't know if I'm missing anything I followed everything exactly. Thank you.

TobKra96, 2020-04-22:
Hi,

sorry for the delay. 
Can you give me some additional info? Does the terminal write something? Does your account have  the rights to execute sudo? Which hardware are you using? Does the webinterface work?
Maybe you can make a video or some pictures. 

Best regards
TobKra

sarahnoreen, 2020-06-01:
Hi! I'm going to hijack this thread as I'm having the same issue. 

For me (using raspberry pi4) the web interface pulls up just fine, and the terminal seems to be responding to the web interface as well as the audio but no LEDs will light up. I can take a few screen shots of whatever you need if that would be helpful.

TobKra96, 2020-06-02:
Hi,
if there is no error in software, maybe there is a problem with the hardware/wiring.
1. Ensure the led strip and the raspberry pi have the same ground connection.
2. Which pin do you use?  The default one is 21. It's the GPIO pin 9.
3. Do you use a resistor between the pin and the Din of the LED Strip?

Maybe you can post the startup log of the program.

sarahnoreen, 2020-06-05:
Ah. Common ground. Likely the issue. I will report back on this. Waiting for some more appropriate sized wires to get in. (My newbie self just learned why to not just run those thing jumper cables from a power supply- oops!)

sarahnoreen, 2020-06-08:
Wiring has been fixed to follow the diagram directly- only difference is it is just a single LED strip instead of 3- numbers have been adjusted in the web app to reflect this. I have attached a file of the beginning of the ran code which includes most of the errors. Let me know if you need further screen shots.
[image]

TobKra96, 2020-06-08:
Hi, the errors of the output come from the audio configuration. Maybe there is some kind of error. Does the program show every 10 seconds an fps counter? There should be fps counters for audio, effects and output. 
Please compare your /usr/share/alsa/alsa.conf with this file. It should look like the same.
[alsa.conf.txt](https://github.com/TobKra96/music_led_strip_control/files/4747590/alsa.conf.txt)

To get the complete log of the starting process you can start the program like this:
 `sudo python3 main.py > log.txt `  
It will redirect your console output to a log.txt file. You can send me the file.

I also draw a little wiring diagram. Please double-check the wires.
[image]

You can test the wiring by executing the examples of rpi_ws281x. 
`/share/rpi_ws281x/python/examples` You have to change the led count and the pin inside the examples.
` sudo nano strandtest.py`
`sudo python3 strandtest.py`

Maybe you can give me this info:
- Do you have a mic?
- Maybe you can take some pics of your wiring?
- Which power supply do you use?

closed 2021-01-03 by TobKra96