TobKra96/music_led_strip_control #21: Web Client only partially reactive

sarahnoreen, 2020-07-12:
Finally got this up and running! However it seems I cant get the web client to be working smoothly. It's basically a 50/50 shot on if changing which mode it's set to will actually work. Sometimes it just doesn't change, other times the effect just freezes and I have to reboot. I'm having particular issues with most of the music reactive modes. Scroll just freezes the current effect every time. Any ideas?

This is running on an rpi 4 and is currently only 30 ws2812's as a prototype.

Thank you!

TobKra96, 2020-07-12:
Hi,
here my ideas:

- Ensure there is no other instance of the program running. You can close all instances by `sudo pklill python3`. This will close all python3 programs.
- Check your configuration file. Ensure you set the right LED_Count, LED_Mid, FPS, and DEVICE_ID. Especially FPS and DEVICE_ID have a big impact on the "smoothness".
- Which Browser do you use? I normally use Chrome. You can try to "hard reload" the page. In Chrome it's pressing F12, then right-click on the reload button or Ctrl+Shift+5.
- Which version of NumPy do you use? I know there is a memory leak in 1.16.
- Maybe you can open htop on a second terminal. Maybe there is a memory leak somewhere.
- Please send me the installed packages with: `sudo pip3 list`

Maybe you can send me the config.json and I can double-check the settings?

Best regards
TobKra

DogWater, 2020-08-06:
points 1-4 check out just fine. (double checked it all and running numpy 1.17.0)

I do see a memory leak- I am unsure how to fix these. It reads out right under the LED Brightness Translated line. `swig/python detected a memory leak of type 'ws2811_t *, no destructor found.'`

pip3 list is
`
pi@raspberrypi:~ $ sudo pip3 list
Package       Version
------------- ---------
asn1crypto    0.24.0
certifi       2018.8.24
chardet       3.0.4
click         7.1.2
cryptography  2.6.1
Cython        0.29.21
entrypoints   0.3
Flask         1.1.2
idna          2.6
itsdangerous  1.1.0
Jinja2        2.11.2
keyring       17.1.1
keyrings.alt  3.1.1
MarkupSafe    1.1.1
numpy         1.17.0
pip           18.1
PyAudio       0.2.11
pycrypto      2.6.1
PyGObject     3.30.4
python-apt    1.8.4.1
pyxdg         0.25
requests      2.21.0
rpi-ws281x    1.0.0
scipy         1.3.0
SecretStorage 2.3.1
setuptools    40.8.0
six           1.12.0
ssh-import-id 5.7
urllib3       1.24.1
Werkzeug      1.0.1
wheel         0.32.3`

So now the non-music effects switch to each other seamlessly. However when I try to switch to a music effect it just freezes for all of them. Terminal reads out that the effect is loaded and gives me effect and output service read outs, but the lights wont move until I switch back to a non-music effect. Very strange. I've tried all the audio settings to no avail.

sarahnoreen, 2020-08-06:
points 1-4 check out just fine. (double checked it all and running numpy 1.17.0)

I do see a memory leak- I am unsure how to fix these. It reads out right under the LED Brightness Translated line. swig/python detected a memory leak of type 'ws2811_t *, no destructor found.'

pip3 list is
`
pi@raspberrypi:~ $ sudo pip3 list
Package Version

asn1crypto 0.24.0
certifi 2018.8.24
chardet 3.0.4
click 7.1.2
cryptography 2.6.1
Cython 0.29.21
entrypoints 0.3
Flask 1.1.2
idna 2.6
itsdangerous 1.1.0
Jinja2 2.11.2
keyring 17.1.1
keyrings.alt 3.1.1
MarkupSafe 1.1.1
numpy 1.17.0
pip 18.1
PyAudio 0.2.11
pycrypto 2.6.1
PyGObject 3.30.4
python-apt 1.8.4.1
pyxdg 0.25
requests 2.21.0
rpi-ws281x 1.0.0
scipy 1.3.0
SecretStorage 2.3.1
setuptools 40.8.0
six 1.12.0
ssh-import-id 5.7
urllib3 1.24.1
Werkzeug 1.0.1
wheel 0.32.3`

So now the non-music effects switch to each other seamlessly. However when I try to switch to a music effect it just freezes for all of them. Terminal reads out that the effect is loaded and gives me effect and output service read outs, but the lights wont move until I switch back to a non-music effect. Very strange. I've tried all the audio settings to no avail.

TobKra96, 2020-08-08:
Hi,

strange behaviour. Can you post your config.json file?
Some effects are not optimized for uneven led counts. Ensure you can devide the led count by 2 and the result should be a full number without ",". Some Matrices calculations are kind of picky.

Which kind of mic do you have? Can you provide me some specs? e.g. Hz, interface (UBS sound card or direkt usb)

Does the terminal show some error messages?
After changing the effect to a music one, does the effect service still outputs the FPS? If not, the effect service crashed.

Best regards,
TobKra

sarahnoreen, 2020-08-18:
Took a note from my buddys post about his error codes and copied over the config.json.back and the effects now work! 

The web interface is still quite a bit behind to update though. Usually have to click the effect then refresh the browser for it to change over. I'm using USB sound card and a cheap lapel mic from amazon. Having a hard time hunting down the specs so that may be a guessing game for me to play with but here are the links in case that helps!

I have been running on input 3- it seems to be slightly more responsive than setting 0 (USB) 

https://www.amazon.com/gp/product/B07PF4N9LM/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B00IRVQ0F8/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1

Thank you for your continued help!

closed 2021-01-03 by TobKra96