TobKra96/music_led_strip_control #23: LED's freeze when using any music effect

rocket770, 2020-09-06:
Hi,

When I ever I try to use a music effect, the LED's completely freeze and become unresponsive. Occasionally, the LED's will work again if I change the mode back to a Non-Music Effect. Everything else is working completely fine just no the Music Effects. 

I'm using a RaspberryPi B+ and I have my config set to the correct microphone (0) and amount of LED's (120).

 This is the output of the terminal
[image]

TobKra96, 2020-09-08:
Hi,

have you already tried to use the sysdefault?
If you followed the installation, you set the Yeti Mic as default.
Maybe try to use the ID 1.

It looks like there is a bug in alsa too:
https://www.raspberrypi.org/forums/viewtopic.php?t=245456
Maybe a `sudo apt-get update` and `sudo apt-get upgrade` help.

Let me know if it worked.

rocket770, 2020-09-09:
Hi,

Switching to sysdefault still gives the same error. 
I've also tried Sudo apt-get update which did actually seem to fix some functionality but the error still applies. 

Another thing that has seemed to help is lowering the sample rate (From 48k to 47k) of the microphone in both the MLSC Config and editing it in the alsa.conf file along with changing some other default settings elsewhere. 

As of now, I still receive the error outputted in the console but functionality wise, it works the vast majority of the time. Occasionally, the LED's will still freeze up but after a re-start of the PI they work pretty well. Ignoring the rare occasion of a crash, I can now still change between audio reactive and regular modes well and they seem to work flawlessly!

Thanks for the help, should be good for now. Appreciate the great software too!

closed 2021-01-03 by TobKra96