TobKra96/music_led_strip_control #26: Add More Options/Colors in Scroll an Config.json

Krashdeez, 2020-11-18:
Is it possible to add more options than low  mid high in scroll effect ? 
That more than 3 colors can be displayed here ? 

Is it also possible to add colors in the "config.json" ? 

Like for example 
 "colors": {
        "red": [255,0,0],
        "Orange": [255,40,0],
        "Yellow":[255,255,0],
        "Green": [0,255,0],
        "Blue": [0,0,255],
        "Light blue": [0,247,161],
        "Purple":[255,0,200],
        "Pink": [255,0,178],
        "White":[255,255,255],
        "Black": [0,0,0],
"Custom1." [1,1,1]
"Custom2." [1,1,1]
"Custom3." [1,1,1]
"Custom4 ": [1,1,1]

Or with 

   "gradients": {
        "Spectral": [[255, 0, 0], [255, 40, 0], [255, 255, 0], [0, 255, 0], [0, 247, 161], [0, 0, 255], [255, 0, 200], [255, 0, 178]],
        "Dance floor": [[255, 0, 0], [255, 0, 178], [255, 0, 200], [0, 0, 255]],
        "Sunset": [[255, 0, 0], [255, 40, 0], [255, 255, 0]],
        "Ocean": [[0, 255, 0], [0, 247, 161], [0, 0, 255]],
        "Jungle": [[0, 255, 0], [255, 0, 0], [255, 40, 0]],
        "Sunny": [[255, 255, 0], [0, 247, 161], [255, 40, 0], [0, 0, 255]],
        "Fruity": [[255, 40, 0], [0, 0, 255]],
        "Peach": [[255, 40, 0], [255, 0, 178]],
        "Rust": [[255, 40, 0], [255, 0, 0]],
        "Safari": [[30, 150, 0], [255, 242, 0], [255, 0, 0]],
        "Jupiter": [[255, 216, 155], [25, 84, 123]],
        "Jamaica": [[255, 255, 0], [0, 247, 161]],
        "Test_Own": [124, 252, 0], [255, 0, 255], [255, 255, 0], [255, 0, 0], [155, 48, 255], [0, 245, 255], [135, 206, 255], [0, 0, 128]]

Krashdeez, 2020-11-18:
Is it also possible to add colors in the "config.json" ? 
- Works - 
i have in 2 lines the , forgotten

Is it possible to add more options than low mid high in scroll effect ?
That more than 3 colors/Options can be displayed here ?

Stil Open, maybe i can add "Spectral" or for Example " Test_Own" as a Color ?

What i Mean, More Options then Low Mid High - 
Example 

-- Low
-- Low-Mid 
-- Mid 
-- Mid-High 
-- High 
-- Maybe "Bass"

TobKra96, 2020-11-24:
Hi,

yes, you can add custom gradients and custom colors.
Add them to the config and the program will show them inside the web interface.

For the additional options, we have to change the scroll effect. We have to define thresholds and add them to the effect.
I will add it to the Feature list of Version 2.0. Maybe we will have a new effect with more Frequencies.

TobKra

TobKra96, 2021-01-01:
Hi, 

I implemented this effect in dev 2.0:

[image]

Thanks for the idea. 👍 
At the moment it's available in the dev_2.0 branch.
I will add some more features and merge them into the trunk.

Have a nice day.

closed 2021-01-03 by TobKra96