TobKra96/music_led_strip_control #27: Website on mobile can't change effect or device settings

itsbrockwillie, 2020-12-10:
I recently setup my WS2812b strip using your controller via a Raspberry Pi 3 model b. This is the second time I've done this and both times it's been pretty straight forward. Everything works great other than not being able to change any effect or device settings on a mobile device. I've tried using Google Chrome and Safari both times to no avail. I even tried using the "Request Desktop Site" and surprisingly it still won't change the effects. When save is pressed and the webpage is reloaded, the effects just revert to their original values. This isn't the biggest problem it would just be nice to change specifically the single color on the fly easily. Thanks!

TobKra96, 2020-12-12:
Hi,
mobile browsers sometimes cache the website and its settings. Maybe you preloaded the page on your smartphone, then change the settings on the PC and the do not reload the page on your smartphone. The old cached setting is still on your phone.
I will change the settings modification to serverside. This is planned for version 2.0.

Workaround:
If you are hard reloading the website after switching the device (you access the webinterface), the configuration should work. I tested this and I am using it very often with my android chrome browser.

Best regards,
TobKra

TobKra96, 2021-01-13:
Improved/Fixed with version 2.0

[image]

closed 2021-01-13 by TobKra96