TobKra96/music_led_strip_control #28: trouble getting to web interface and leds wont turn on

Electricmitch, 2020-12-20:
I dont know how to get to web interface and when I run program leds dont light up and it says this: 

[image]
[image]
https://mail.google.com/mail/u/0?ui=2&ik=fe7cad4fdb&attid=0.5&permmsgid=msg-f:1686580955660738307&th=1767f0717b18df03&view=att&disp=safe

also does this wiring look right?
[image]

TobKra96, 2020-12-23:
Do you have another program, that blocks port 80? 

The default pin config is GPIO Pin 21. See:

[image]

Electricmitch, 2021-01-02:
Where is port 80 on the Rasberry pi? I only saw that it goes up to 40. Also how do you access the web interface?

TobKra96, 2021-01-02:
Hi,

with port 80, I mean the software/network port. https://en.wikipedia.org/wiki/Port_(computer_networking)
The default port for Web pages is 80.

You can reach the Webinterface, by connection via a web-browser. The device with the web-browser have to be in the same network as the raspberry pi. http://<raspberry-pi-ip>
Replace "<raspberry-pi-ip>" with the IP of your raspberry pi. You can find the IP inside the console.
In your case it should be http://192.168.0.13 .

I marked the physical connection of the led strip on the post before. GPIO 21 or PIN 40.

Have a nice day.

Electricmitch, 2021-01-02:
I was able to access the web interface but how do I check to see if port 80 is available and/or working correctly?

Electricmitch, 2021-01-02:
Also it keeps saying “jack server is not running or cannot be started” how do I fix this?

TobKra96, 2021-01-03:
Hi,

if you can access the web interface, the port 80 work.
You can try to select a none audio effect to test the led strip output.

Now, the Jack Server is your audio server, which should be provided by your OS. 
Have you installed:
`sudo apt-get install libatlas-base-dev`
`sudo apt-get install portaudio19-dev`

Have you configured the `/etc/asound.conf` and `/usr/share/alsa/alsa.conf` like I described in the installation guide?
Can you post both files?

Installed Alsa packages can be shown by:
`dpkg -l |grep alsa`
You should get at least: alsa-utils

With the following command, you can check the installed drivers:
`lsmod`

There should be a line with:
snd_usb_audio and snd_bcm2835

You can reinitialize the audio config by:
`alsactl init`
Maybe you have to configure `/etc/asound.conf` and `/usr/share/alsa/alsa.conf` again.

You can try to access the alsamixer: `alsamixer -c 1` with F5 you can see all devices.

Try to record a audio file with: `arecord --device=hw:1,0 --format S16_LE --rate 44100 -c1 test.wav -V mono`
Do you hear some sound?

Have a nice day

Electricmitch, 2021-01-06:
Ok thank you so much for all the advice man! I really appreciate it! And I will work on those steps above but right now my power supply section is being a bit tricky. So when I don’t even have the data pin connected to anything, when I connect my positive and negative leads of my power supply to the LED strip, and turn the voltage dial up to 5, the current starts to go up as well, even though no data is going into the lights. This is concerning because I don’t want to blow the lights. Can you help me figure out why this is happening?

Electricmitch, 2021-01-06:
[image]
This is how my leads are hooked up to my power supply

TobKra96, 2021-01-06:
Hi,

some ideas:
- Check that the cables at end of the LED Strip do not touch each other.
- The microcontroller, that controls the led will need some power, even if there is no data signal. How many LEDs do you have and how high is the current? ~60mA per LED is the max power usage for a WS2812B LED. On idle I would say 5mA for each LED would be normal. Which kind of LED strip do you use? WS2812B?
- Please ensure your power supply is big enough.

Please write the specs of your setup.
Power Supply Voltage, Max Amp. LED Stripe Model (WS2812B or other), LED count, led/meter (maybe you have to connect the end of the led too)

Have a great day.

Electricmitch, 2021-01-08:
Good evening,

I checked and the wires on the end are not touching each other. Specs:
Power supply: 30v, 10Amps
LEDs: WS2812b, 300 LED count, 16.4 feet, 60 LEDs per meter

TobKra96, 2021-01-09:
Is it a laboratory power supply?
30 V is way too high. You need a 5V Power Supply for the ws2812B LEDs.
For 300 LEDs you will need at least 300*60mA = 18A. I would use a 20A psu.
If you are using a laboratory power supply it should go into the current limiter if the led strip will need more than 10 A. This won't break the led strip.

Electricmitch, 2021-01-09:
So it is an adjustable power supply so I am only turning the voltage knob
to 5v. But I was wondering how to properly ground everything with the
raspberry pi

TobKra96, 2021-01-09:
I made a wiring schematic for the ws2815. Maybe this will help you. The wiring should be the same (no Bi pin), but the specs will be different. Gnd should be connected with the led strip, the raspberry pi and the power supply.
[image]

TobKra96, 2021-01-31:
I think it worked for you. Closed

closed 2021-01-31 by TobKra96