TobKra96/music_led_strip_control #29: Music effects don't work

alexdobo, 2020-12-23:
I'm unable to get any music effects to work
I can control the LEDs with the non music effects
The sound card works when I use aplay and arecord to test it

I get the following error when running main.py:
```
pi@raspberrypi:~/share/music_led_strip_control/server $ sudo python3 main.py
Init the programm...
Settings loaded
Starting Output component..
Start Effects component...
LED Brightness: 10
LED Brightness Translated: 25
Output component started.
NotificationService component started.
Init finished
Programm started...
 * Serving Flask app "libs.webserver" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Server listen to 192.168.1.102:65432
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Process Process-6:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/share/music_led_strip_control/server/libs/audio_process_service.py", line 33, in start
    self._default_device_id = self._py_audio.get_default_input_device_info()['index']
  File "/usr/local/lib/python3.7/dist-packages/pyaudio.py", line 949, in get_default_input_device_info
    device_index = pa.get_default_input_device()
OSError: No Default Input Device Available
Effects component started.
```

The card ID I have selected is 1
As per arecord -l:
```
pi@raspberrypi:~/share/music_led_strip_control/server $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~/share/music_led_strip_control/server $
```

I have changed /usr/share/alsa/alsa.conf to 
```
defaults.ctl.card 1
defaults.pcm.card 1
```

I have changed ~/.asoundrc and /etc/asound.conf to
```
pcm.!default {
  type hw
  card 1
}

ctl.!default {
    type hw
    card 1
}
```

In the Audio Source ID setting, I have set it to 1 (using chrome).

Any ideas what I could try next?

Thanks in advanced

TobKra96, 2020-12-23:
Hi,

it looks like pyaudio cannot get the default audio device. Have you installed `sudo apt-get install portaudio19-dev` ? This is the audio driver.
Does the arecord record something? Does the mic work?

alexdobo, 2020-12-24:
Portaudio19-dev is installed

arecord does record things. I was able to record audio and play it back.

TobKra96, 2020-12-25:
Which os do you use?
Ubuntu need: `sudo apt-get install python-gnuradio-audio-portaudio`

You can check if pyaudio works by execute the following python commands:
>>> import pyaudio
>>> pa = pyaudio.PyAudio()
>>> pa.get_default_input_device_info()

Maybe you need : `sudo apt-get install libasound-dev` too.

alexdobo, 2020-12-26:
I'm running Raspbian 10 (Buster) on a RPi Zero W

I ran that python, here's the output:
```
pi@raspberrypi:~ $ sudo python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyaudio
>>> pa = pyaudio.PyAudio()
>>> pa.get_default_input_device_info()
{'index': 7, 'structVersion': 2, 'name': 'default', 'hostApi': 0, 'maxInputChannels': 1, 'maxOutputChannels': 2, 'defaultLowInputLatency': 0.008684807256235827, 'defaultLowOutputLatency': 0.008707482993197279, 'defaultHighInputLatency': 0.034829931972789115, 'defaultHighOutputLatency': 0.034829931972789115, 'defaultSampleRate': 44100.0}
>>>
```
I've been able to clear up all of the unknown PCM stuff by following [this stackoverflow answer](https://stackoverflow.com/questions/7088672/pyaudio-working-but-spits-out-error-messages-each-time)

I tried running it again today, and I get a different error, but it still doesn't work
```
pi@raspberrypi:~/share/music_led_strip_control/server $ sudo python3 main.py
Init the programm...
Settings loaded
Starting Output component..
Start Effects component...
NotificationService component started.
LED Brightness: 10
LED Brightness Translated: 25
Output component started.
 * Serving Flask app "libs.webserver" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Init finished
Programm started...
Server listen to 192.168.1.102:65432
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
Found the following audio sources:
1 - sysdefault - 48000.0
Selected ID: 1
Use 1 - sysdefault - 48000.0
Start open Audio stream
Effects component started.
Set effect to: effect_scroll
Save settings
192.168.1.113 - - [26/Dec/2020 02:16:08] "POST /setActiveEffect HTTP/1.1" 200 -
New effect found: EffectsEnum.effect_scroll
Reload config..
Refresh Output...
Settings loaded
Refresh effects...
Settings loaded
LED Brightness: 10
LED Brightness Translated: 25
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
Output refreshed.
Effects refreshed.
Config reloaded.
Output Service | FPS: 66.56040625247957
Effect Service | FPS: 69.00909853732375
```

Here's the error:
```
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
```

TobKra96, 2020-12-27:
Hi,
that's a bug of the https://github.com/jgarff/rpi_ws281x library, but it should work anyway.
Does the strip work? Do you see any light if you are using the none music effects?
- Oh, I see you have already written it in the first post, yes none music effect works.

Do you see any kind of reaction, if you hit the microphone with your fingers? Maybe the mic level is extremely low?

You selected the sysdefault. Maybe the link is wrong.
I never used the zero pi before. It looks like, it will need some additional settings. Try this:
https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/updating-alsa-config

Please let me know if it works :)
Have a nice day.

alexdobo, 2020-12-28:
I don't get any kind of reaction at all

I've noticed that sometimes after I end the program, and start it up again, the sound card goes missing for some reason?

The more I look into this, I think that it isn't a problem with your code, but some kind of problem with the RPi or linux

I (think I) have gotten all of the ALSA configs correct, as the microphone and speakers work using aplay and arecord

TobKra96, 2020-12-28:
Hi,

we can try to test the audio lib, that I am using "pyaudio".
This is an example code, to record a file with pyaudio.
https://gist.github.com/TobKra96/67ba2469f45d991bd331570edc738c73

This script should record 5 seconds. Modify the device index. If you do not hear some noise, when you play the recorded file, there will be some issues with the OS config.

Have a nice day.

alexdobo, 2021-01-01:
My RPi overheated and died today
I'm going to assume that it was something wrong with my RPi and close this
Thanks for your help

closed 2021-01-01 by alexdobo