TobKra96/music_led_strip_control #36: can you mabey make a little video

rickblij, 2021-01-18:
Open the webinterface of MLSC Search inside your Browser for the hostname of the raspberry pi, or the ip.
Navigate to "Edit General Settings/Audio Setting".
There should be an input filed to enter the "Audio Source ID".
Press "Save".

Now we change the led stip size, to match it with your strip.
Go to "Edit General Settings/Device Settings".
Enter the count of LEDs you use inside "Numbers of LEDs".
Change the mid of your led strip. This will affect the mirror option of many effects.
You can use the half of your "Numbers of LEDs" or set a custom number, if you got a corner you want to start the mirror effect. You will get the best results with the half of "Numbers of LEDs".
Press "Save".

Restart the programm and have fun with MLSC.

TobKra96, 2021-01-21:
Sorry, at the moment I am in the development of the next version.
Such a video would take a lot of time. Can you please describe your problem. Or where you stuck.
For the next version I will record a complete tutorial, but not for the old version.

rickblij, 2021-01-23:
Now my problem is that I can’t see my http://myipraspberrypi:10000 I can’t find this it doesn’t work and when comes you new video ?

TobKra96, 2021-01-23:
The default port web interface port is 80. If you did not change it and no other program blocks it, the web interface will be available with http://<hostname>.
If you cannot reach the raspberry pi by host name, try the Ip address. http://<IP>.

At the moment I am preparing the Installation Guide for the new Version. I think the video will be online in two weeks.

TobKra96, 2021-01-31:
It looks like, the issue can be closed.

closed 2021-01-31 by TobKra96