TobKra96/music_led_strip_control #45: I can’t get it

rickblij, 2021-01-26:
[image]
[image]
How can I fix it

TobKra96, 2021-01-26:
The default port of webmin is 10000. Please try http://raspberrypi:10000

rickblij, 2021-01-26:
[image]
Still nothing

TobKra96, 2021-01-26:
Execute `/etc/init.d/webmin restart`

rickblij, 2021-01-26:
Are you on now mabey you can do it on TeamViewer

TobKra96, 2021-01-26:
Yes I am online now.  Click on details and accept.

rickblij, 2021-01-26:
[image]
There is nothing

TobKra96, 2021-01-26:
ok, I can try to take look on it with team viewer.

rickblij, 2021-01-26:
How can I download that in a raspberry pi

rickblij, 2021-01-26:
[image]

TobKra96, 2021-01-26:
Do you have discord? Here ist the invite link: https://discord.gg/D8pdQpAr

TobKra96, 2021-01-31:
Topic closed. I tried to help you in discord.

closed 2021-01-31 by TobKra96