TobKra96/music_led_strip_control #46: update

dkcandy, 2021-01-27:
Hi,

Could you please help how can i upgrade to the latest version?

Thanks

TobKra96, 2021-01-27:
Hi,

you have to reset the local changes of the checkout repository and pull the newest changes:

Go inside your repository folder on the raspberry pi (with e.g. cd `/share/music_led_strip_control`)
`sudo git fetch --all`
`sudo git reset --hard origin/master`
`sudo git pull origin master`

After this, you can start with the configuration: https://github.com/TobKra96/music_led_strip_control/wiki/Installation-Guide#iv-configure-music-led-strip-control

dkcandy, 2021-01-27:
it works perfectly, thank you very much for your work

TobKra96, 2021-01-27:
Have fun with the project. :smiley:

closed 2021-01-27 by TobKra96