TobKra96/music_led_strip_control #48: New Feature: Ability to create own Programs and to trigger certain modes per url

Loxitation, 2021-02-04:
On another LED-Server I used till today I was able to create own effects / programs for the stripe, and to trigger them via php-files. Is there a possibility here too? Would be really important because I've built my DJ-Logo with reactive-LED's and want twitch viewers to be able to trigger cmds to control the leds.

:) But anyway, verry verry cool tool!

TobKra96, 2021-02-05:
Hi, yes you can change effect via a rest api call. I will give you an example code this evening.

Loxitation, 2021-02-05:
Thanks a lot! :-)

TobKra96, 2021-02-05:
Hi, here is the example code for changing an effect via PHP:
https://gist.github.com/TobKra96/7004030cfb846f9792b3ec2afe423275

Replace <raspberry-pi-ip> with the IP/hostname of your raspberry pi.
There are two variants to change an effect. You can change it for one specific device or for all.

You can set the effect by setting the effect id into the array:
$dataAllDevices = array('effect' => "effect_gradient");
You can find all effects ids here: https://github.com/TobKra96/music_led_strip_control/blob/master/server/libs/effects_enum.py
For the single device command, you will need to add the device id. You can find the device ID inside your config.json file in the following folder: music_led_strip_control/server/libs/config.json. 

So you want to create your own effect? 
I can make a small guide for the wiki if you want. 

Have a great day. :)

Loxitation, 2021-02-05:
Thanks a lot! :-)

The Tutorial would be perfect, because my LED's are arranged in 2 triangles, so for them the effects need to look a bit differend then for leds that are in one line. Would be a great opportunity to individualize even further! :-)

TobKra96, 2021-02-06:
Hi,

here is the link to the guide. [Wiki](https://github.com/TobKra96/music_led_strip_control/wiki/Create-a-custom-effect)
Let me know if you need help or there is a part in the guide, which is not clear. 

Have a great day :)

Loxitation, 2021-02-07:
Hey, I've tested to trigger the effect per php, but it does not seem to get triggered right. What do you mean with getinfo.php? I haven't found such file and also don't know what I should put in there :D

TobKra96, 2021-02-08:
Ups, that was a comment for another topic. Ignore the comment.
Do you get an error message?
Or do you get an output on the site like: "Send All Request"?
How do you start the PHP script?

Loxitation, 2021-02-08:
I am currently calling the php file via browser, so I enter the url to the .php and get "Send all Request" as echo, but nothing else happens. I already tried to debug the script to add an errorhandler and more echos true the process, but it just keeps responding "send all Request". :(

TobKra96, 2021-02-08:
Have you replaced the IP in the lines:
`curl_setopt($ch, CURLOPT_URL, "http://<raspberry-pi-ip>/SetActiveEffect");` and `curl_setopt($ch, CURLOPT_URL, "http://<raspberry-pi-ip>/SetActiveEffectForAll");`

ensure you can reach the raspberry pi from the location you start the PHP script (location of the PHP server).
You can try to echo the result of the request.

Loxitation, 2021-02-08:
Sure, replaced both ip's, and the pi is reachable from my pc. Can ping, can execute other php files in the same directory. I tried to echo the result, but get no result.

Loxitation, 2021-02-08:
Getting an 500 internal server error in the f12 debug console of chrome and firefox

TobKra96, 2021-02-08:
Ok, some new ideas and questions:
Do you use the newest version?
Can you starte mlsc manually? Kill all instances with `sudo pkill pyhton3` and start the main.py in /share/music_led_strip_control/server with `sudo python3 main.py`
Now try to execute the php script and look inside the terminal output. Post it here.
Have you changed the default webport?
Can you enter the webinterface?

Loxitation, 2021-02-08:
Newest Version: Yes
Running manually: Yes
Terminal Output after executing Script: Nothing
Console Output in Webbrowser: 500 Internal Server Error
Default Webport: Changed to 8080 because pihole is already occupying 80
Webinterface accessable: Yes
Changed URL in php-file to 192.168.178.53:8080 : Still nothing in console when running
Changed everything back.
Added `curl_setopt($ch, CURLOPT_PORT, 8080);`: Nothing
Modified `$ch = curl_init();` to `$ch = curl_init('http://192.168.178.53:8080');`: Nothing
Tried with http, https and nothing: Nothing
Changed `curl_setopt($ch, CURLOPT_HTTPHEADER, array('Content-Type:application/json'));` to `curl_setopt($ch, CURLOPT_HTTPHEADER, array('Content-Type: application/json; charset=utf-8'));` : Nothing

Built in some echos for debugging: The `$data` says in line 19 `{"effect":"effect_gradient"}`. Then comes `$ch = curl_init();` and after that no further response.

Seems a bit lile curl is not working correctly,  but already checket that its installed and working. I will now try to rebuild the script with fsock instead of curl.

TobKra96, 2021-02-08:
Hi,

you can check if curl is installed with:
Create a new file with the following content:
`<?php phpinfo(); ?>`

You should see this chapter:
[image]

Activate curl: https://stackoverflow.com/questions/4477535/curl-init-function-not-working

I used PHP Version 8.0.1. for the tests. Which one do you use?

If everything works, you should see something like this inside the terminal:
`x.x.x.x - - [08/Feb/2021 16:43:09] "POST /SetActiveEffectForAll HTTP/1.1" 200 -`

Loxitation, 2021-02-08:
Curl indeed does not seem to be activated, tho I removed the ; in the php.ini file. Strange. I use php7.3.19

Loxitation, 2021-02-08:
Got the solution. curl was corrupt. -.- Sorry for bothering you with always those dummie stuff.

closed 2021-02-08 by Loxitation

TobKra96, 2021-02-09:
I am happy, that it worked. :-)
Let me know if you have questions about the custom effect.

Loxitation, 2021-02-12:
Hey, one further question I need to ask: Is there a way to adress leds by their number? So lets say I want to color 0-70 red, 71-110 blue and so on?

TobKra96, 2021-02-12:
Hi, no at the moment there is no effect like this.
It is already on the to-do list ;) #56

Loxitation, 2021-02-12:
Alright, thank you! :-)

Loxitation, 2021-02-13:
Hey, me again, your favorite annoying question-asker :D 

I got some spare time today and started doing custom effects. Is there a way to hardcode the colors? For example a strobe effect just needs to be white. In the wiki it says `output[0][:] the [0] represents the red color channel. [:] selects all LEDs. ` 

So output_array[0][:][255] should trigger the red leds in full power or am I wrong? Also tried output_array[0][:] = "255"[0], but doesnt seem to work either.

TobKra96, 2021-02-13:
Hi,

it could look like this:

```python
output_array[0][:] = 255
```

You will set the [0] dimension, which is red. Select all LEDs with [:]. and set them to all to 255

Loxitation, 2021-02-14:
So you've saved me from all my problems. Only one further question: When I would like to update, does the program recognize my custom effects, or do I need to restore them manually?

TobKra96, 2021-02-14:
The update script will overwrite your changes. You can try to merge the master with your changes. Maybe you have to resolve conflicts, that are more challenging than paste your custom effect into the new version.
So do a backup, before updating/merging.

Loxitation, 2021-02-14:
Okay, thanks :)