TobKra96/music_led_strip_control #49: Issue

Loxitation, 2021-02-04:
`> Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/s2lled/music_led_strip_control/server/libs/output_service.py", line 51, in start
    self._current_output = self._available_outputs[current_output_enum](self._device)
  File "/home/pi/s2lled/music_led_strip_control/server/libs/outputs/output_raspi.py", line 46, in __init__
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)
`

Also, I've installed numpy1.20.0, but it says numpy not found. Can you help me with that issue too? When I do python main.py it starts with the above error, when I say python3 main.py it says numpy not found. Already tried to reinstall numpy 3 times ._.

TobKra96, 2021-02-05:
Hi, it looks like Python 3 doesn't know numpy.
Have you installed numpy with the `pip3`command?
`sudo pip3 install -I numpy==1.17.0`
You can see all installed packages with
`sudo pip3 list`

Loxitation, 2021-02-05:
Checked that, pip3 list finds numpy, got 1.20.0, downgraded to 1.17.0, same issue. Will try a fresh clean install on a clean pi and report back in the evening wether that worked or not.

Teraskull, 2021-02-06:
@Loxitation Can you try to install the following:

```bash
sudo apt-get install python3-numpy python3-scipy
```

And then retry running `main.py`?

Loxitation, 2021-02-06:
@Teraskull tried that just now, same issue. Sorry for the late reply, haven't got earlier to testing again.

@TobKra96 The clean install also throws the same error, no module named numpy. Error is still the following:

pi@raspberrypi:~/s2lled/music_led_strip_control/server $ sudo su
root@raspberrypi:/home/pi/s2lled/music_led_strip_control/server# python3 main.py
Traceback (most recent call last):
  File "main.py", line 8, in <module>
    from libs.device_manager import DeviceManager
  File "/home/pi/s2lled/music_led_strip_control/server/libs/device_manager.py",                                                                                                                                                              line 1, in <module>
    from libs.device import Device # pylint: disable=E0611, E0401
  File "/home/pi/s2lled/music_led_strip_control/server/libs/device.py", line 3,                                                                                                                                                              in <module>
    from libs.effect_service import EffectService
  File "/home/pi/s2lled/music_led_strip_control/server/libs/effect_service.py",                                                                                                                                                              line 5, in <module>
    from libs.effects.effect_bars import EffectBars # pylint: disable=E0611, E04                                                                                                                                                             01
  File "/home/pi/s2lled/music_led_strip_control/server/libs/effects/effect_bars.                                                                                                                                                             py", line 1, in <module>
    from libs.effects.effect import Effect # pylint: disable=E0611, E0401
  File "/home/pi/s2lled/music_led_strip_control/server/libs/effects/effect.py",                                                                                                                                                              line 1, in <module>
    from libs.color_service import ColorService # pylint: disable=E0611, E0401
  File "/home/pi/s2lled/music_led_strip_control/server/libs/color_service.py", l                                                                                                                                                             ine 1, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

TobKra96, 2021-02-06:
Can you check your installed python version?
`sudo python3 --version`
The version should be Python 3.7.3.
Try `sudo python3.7 main.py`.

Loxitation, 2021-02-06:
pi@raspberrypi:~ $ sudo python3 --version
Python 3.7.0

@TobKra96 saved my night. pip3 was not linked to python3 as it seems. Sorry for bothering you with such stuff, still learning :D Thanks you verry verry much for your help! :)

Loxitation, 2021-02-06:
So for all out there, to sum it up, @TobKra96 s last answer solved the issue. Pip3 was not linked to python3.

`sudo python3 -m pip install -I numpy==1.17.0 `

solved the issue.

closed 2021-02-06 by Loxitation