TobKra96/music_led_strip_control #50: Client doesn't work in Firefox

paulheald, 2021-02-04:
When you save the settings for the device in Firefox, the browser rejects it with NS_BINDING_ABORTED. I looked at this quickly, I think it's because the page reloads faster than the post to the rest endpoint, but I'm not 100% sure. It works for me in Chrome with no issues.

TobKra96, 2021-02-05:
Hi, thanks for the debugging. 👍 I will try to fix it this evening.

paulheald, 2021-02-05:
Great! I also noticed it does the same on FF on iOS. Let me know if I can help. I have only just started digging into the code. Loving this project! :)

TobKra96, 2021-02-05:
Hi, I think you saw the right problem. The Websites reloads before the ajax requests were finished.
Now the reload should wait until all requests are sent.
Let me know if you see any more problems.

Have a great day.

paulheald, 2021-02-06:
Pulled and it works perfectly on both phone and desktop. Thank you!

closed 2021-02-07 by TobKra96