TobKra96/music_led_strip_control #51: rpi_ws281x is available as a pip package

Teraskull, 2021-02-06:
The `rpi_ws281x` module can be installed via pip with no issues. Tested on Raspberry Pi 3B+.
I skipped the build process completely and everything works.
```
sudo pip3 install rpi_ws281x
```

Also a nice idea would be to make a bash script to automatically install all required packages, so that we don't need to copy paste each line separately.

Something like this:

`mlsc_setup.sh`
```bash
# Setup script for MLSC
# https://github.com/TobKra96/music_led_strip_control

# Update the libraries:
sudo apt-get update
sudo apt-get upgrade

# Install Audio Driver:
sudo apt-get -y install libatlas-base-dev portaudio19-dev

# Install Python:
sudo apt-get -y install python3

# Install Pip.
sudo apt-get -y install python3-pip

# Install all Python libraries:
sudo pip3 install --upgrade pip --yes     # Upgrade Pip to the newest version.
sudo pip3 install -I numpy==1.17.0 --yes  # Offers a lot of mathematic functions and matrix manipulation. We need this version because 1.16 has a memory leak by using queues.
sudo pip3 install rpi_ws281x --yes        # Raspberry Pi PWM library for WS281X LEDs.
sudo pip3 install flask --yes             # The webserver component.
sudo pip3 install pyaudio --yes           # Offer the audio input stream, which will be processed.
sudo pip3 install scipy==1.3.0 --yes      # Offers a gaussian filter.

sudo apt install -y python3-scipy         # Fallback scipy module if the previous one fails to install.
```

TobKra96, 2021-02-06:
Hi,
thanks for the hints and new feature ideas. I will collect them for the next version.
👍

Teraskull, 2021-02-06:
I am currently working on the complete setup script, will make a Pull Request once done.

TobKra96, 2021-02-06:
Wow, that's perfect :) Thank you.

TobKra96, 2021-02-11:
Setup script implements the package. See #54
Thanks to @Teraskull

closed 2021-02-11 by TobKra96