TobKra96/music_led_strip_control #52: Color presets are not correct when using a GRB strip instead of an RGB one

Teraskull, 2021-02-06:
When I select the `Red` color preset, the strip lights green, and vice versa.

As an option, a toggle can be added to the Device Settings, to select between RGB/GRB strips.
But then the color presets have to be refactored, because they are hard-coded right now.

TobKra96, 2021-02-06:
Hi, do you use the esp client or the direct output via raspberry pi?

**Raspberry Pi Output**
With the raspberry pi. You can set the following setting in the file:
https://github.com/TobKra96/music_led_strip_control/blob/master/server/libs/outputs/output_raspi.py

In line 33. Between `self.channel = ws.ws2811_channel_get(self._leds, 0)` and `ws.ws2811_channel_t_count_set(self.channel, self._led_count)`

```
#LED_STRIP      = ws.WS2811_STRIP_RGB
LED_STRIP      = ws.WS2811_STRIP_GBR
#LED_STRIP      = ws.SK6812_STRIP_RGBW
#LED_STRIP      = ws.SK6812W_STRIP

ws.ws2811_channel_t_strip_type_set(channel, LED_STRIP)
```

**ESP Output**
For esp you have to change the following in the file: https://github.com/TobKra96/music_led_strip_control/blob/master/client/udp/esp/nodemcu_client/nodemcu_client.ino

Line 23: `NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> ledstrip(NUM_LEDS, PixelPin);`
Change NeoGrbFeature to the wanted output type:
https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object
NeoGrbFeature
NeoGrbwFeature
NeoRgbFeature
NeoRgbwFeature
NeoBrgFeature
NeoRbgFeature

Let me know if it worked for you. I will add a selection of the Strip Types in the next version.
Have a great day :)

Teraskull, 2021-02-06:
I only used one Raspberry Pi 3B+ for both server and client. I will check if changing the output helps, thanks.

Teraskull, 2021-02-07:
Thanks, the proposed code works:

```python
LED_STRIP = ws.WS2811_STRIP_GBR
ws.ws2811_channel_t_strip_type_set(self.channel, LED_STRIP)
```

It's then really easy to add a similar toggle to the settings, like the `Used Output Type` dropdown.

TobKra96, 2021-02-07:
Yes, that's the plan. I will implement it after the pull request is done :)
Have a great day!

TobKra96, 2021-02-14:
Added Dropdown for Raspberry pi Output:
[image]

I accidentally used the master. Sorry

Teraskull, 2021-02-14:
No problem. It is missing the `ws.WS2812_STRIP`, which is the default GRB WS2812B Strip.

TobKra96, 2021-02-14:
Fixed it: 
[image]

Teraskull, 2021-02-14:
Great, we can close this now.

closed 2021-02-14 by Teraskull