TobKra96/music_led_strip_control #53: Color does not get updated when clicking "Save" more than once

Teraskull, 2021-02-06:
I am running MLSC via a systemd service.

When I edit the color of, for example, `Single` mode, and click "Save", the color gets updated. If I try to change it to another color, nothing happens when clicking "Save".

It only changes when restarting the whole service.

TobKra96, 2021-02-06:
Hi,
this sounds like a bug. I think the program crashes after setting the color.
Kill all instances with `sudo pkill python3`.
Now start the program manually. Change to the directory you have installed mlsc. E.g:
`cd /share/music_led_strip_control/server`
Start the program with `sudo python3 main.py`.
Set the color in the web interface and look inside the terminal.
Copy the exception and paste it here. Maybe we can fix the problem with the info inside the exception. :)

Have a great day!

Teraskull, 2021-02-06:
I will check if it throws an exception, thanks.

By the way, do you mind if I clean up the whole codebase to PEP standards? It has a lot of trailing white-spaces and inconsistent indentations. It won't break anything, just will be easier to read.

TobKra96, 2021-02-06:
Yes, that would be nice. Sometimes I forget the python code naming convention. My main programming language is C#.

Teraskull, 2021-02-06:
No problem, your project is the best from all alternatives, trust me, I tried a lot of them :D

Teraskull, 2021-02-07:
There is no exception, but the output after the second "Save" is:

```bash
Save settings
Save settings
Preparing new Notification...
Notification Item prepared: NotificationEnum.config_refresh all_devices
Notification Item put into queue.
192.168.1.217 - - [07/Feb/2021 03:22:20] "POST /SetEffectSettingForAll HTTP/1.1" 200 -
Preparing new Notification...
Notification Item prepared: NotificationEnum.config_refresh all_devices
Notification Item put into queue.
192.168.1.217 - - [07/Feb/2021 03:22:20] "POST /SetEffectSettingForAll HTTP/1.1" 200 -
Effect Service | FPS: 58.29875599416221 | Device: Raspberry Pi
Output Service | FPS: 59.5612610053962 | Device: Raspberry Pi
```

TobKra96, 2021-02-07:
I could not reproduce it here. 
It looks like the notification service does not get the message, that there is a reload.

It should look something like this:
```
Save settings
Save settings
Preparing new Notification...
Notification Item prepared: NotificationEnum.config_refresh all_devices
Notification Item put into queue.
192.168.178.86 - - [07/Feb/2021 13:03:51] "POST /SetEffectSettingForAll HTTP/1.1" 200 -
Preparing new Notification...
Notification Item prepared: NotificationEnum.config_refresh all_devices
Notification Item put into queue.
192.168.178.86 - - [07/Feb/2021 13:03:51] "POST /SetEffectSettingForAll HTTP/1.1" 200 -
NotificationService: New Notification detected.
Reloading config...
Device Manager received new notification: NotificationEnum.process_pause - all_devices
Device Manager received new notification: NotificationEnum.config_refresh - all_devices
Device count before: 4
Entering reload_config()
Settings loaded
Leaving reload_config()
Device count after: 4
Restarting device_0
Refreshing config of device: Cabinet small
Stopping device: Cabinet small
Starting device: Cabinet small
Restarted device_0
Restarting device_1
Refreshing config of device: Kitchen
Stopping device: Kitchen
Starting Output service... Device: Cabinet small
Found output: OutputsEnum.output_udp
Output component started. Device: Cabinet small
Starting Effect Service component from device: Cabinet small
Starting device: Kitchen
Settings loaded

```

Which browser do you use?
Do you see the "Device Manager update" after the saving: `Device Manager | FPS: 139.20690341851974`
Do you see the "Audio Service update" after the saving? `Audio Service Callback | FPS: 91.79515013569115` and `Audio Service Routine | FPS: 135.43556459685493`

If you close the program with "ctrl+c" you will get the last traceback. Maybe we can find here the line, which will cause the hang-up. Copy and paste all lines in here, after pressing "ctrl+c".

Have a great day!

Teraskull, 2021-02-08:
Bug was on my end. I did not connect a microphone while testing Non-Music Effects. After connecting the microphone, the "Save" button works as expected.

TobKra96, 2021-02-09:
I will reopen the issue because the mlsc should work even without a mic. Thank you for the debugging.

rocket770, 2021-02-13:
Hi,

I get the same issue where I can only edit the settings once before they stop taking effect. Restarting the pi will make the changes work but you can only edit one setting afterwards and have it work instantly. I am not using a microphone but even with it connected I still get the same issue. This is only a recent issue and I have tried re-downloading the software.

Everything else works great :)

Teraskull, 2021-02-13:
@rocket770 did you properly assign the microphone in the General Settings? Are there any errors logged in the terminal?

rocket770, 2021-02-13:
When I do use the microphone, the audio effects for it work fine. There are no other errors when I use it. I still get the saving issue with or without it.

Teraskull, 2021-02-13:
Hmm, weird. We are still looking into this issue, but I "fixed" it by connecting the microphone, rebooting Raspberry Pi, and inputting Audio Source ID `2` in the General Settings. Since then I couldn't replicate the bug, which makes it harder to track.

TobKra96, 2021-02-14:
I saw some Issues, that could be caused by the missing mic. Please let me know if you see the "save" issue with the newest version.
Have a great day!

TobKra96, 2021-02-19:
I will close this issue. If you see the error again, please open it or create a new issue.

closed 2021-02-19 by TobKra96