TobKra96/music_led_strip_control #55: MLSC autostart

BomBastikDE, 2021-02-06:
Thx for this great Project. Everything works well - but the autostart doesn't work.
i have done: "v. Add MLSC to the autostart" from wiki - but wenn i reboot the MLSC doesn't start - I have to start ist with "sudo /etc/init.d/mlsc start" - the it works - can you help?

Teraskull, 2021-02-07:
I am using `systemd` for autostart, it is working well.

Try the following steps:

1. Create a file `mlsc.service` in `/etc/systemd/system/`:

```bash
sudo touch /etc/systemd/system/mlsc.service
```

2. Add this to the created file:
```ini
[Unit]
Description=Music LED Strip Control
After=network.target

[Service]
Type=simple
User=pi
WorkingDirectory=/share/music_led_strip_control/
ExecStart=sudo python3 server/main.py
KillMode=process
Restart=always
RestartSec=5
KillMode=control-group

[Install]
WantedBy=multi-user.target
```

3. Next, run these commands to enable and start MLSC:

```bash
sudo systemctl daemon-reload

sudo systemctl enable mlsc.service

sudo systemctl start mlsc.service
```

4. Reboot:

```bash
sudo reboot
```

5. Check if the service is running:

```bash
sudo systemctl status mlsc.service
```

6. Done! Should restart on reboot.

P.S. The `WorkingDirectory` specified here is in `/share`, so if you did not change anything during the installation, it should work without issues.

@TobKra96, can you please add these instructions to the wiki? Thanks!

TobKra96, 2021-02-07:
Hi,
yes, I will do it. 
For the ESP + Access Point solution, I have to find a way to wait for the access point.
It has to be initialized first before we start the mlsc program.
The current solution uses a "dirty sleep". See: [Installation Guide](https://github.com/TobKra96/music_led_strip_control/wiki/Installation-Guide#setup-the-raspberry-pi)

Have a great day!

BomBastikDE, 2021-02-07:
Thx - now it works

closed 2021-02-07 by BomBastikDE