TobKra96/music_led_strip_control #57: New logo propositions

Teraskull, 2021-02-08:


Teraskull, 2021-02-08:
## Spectrum Logo

Option 1:
[image]

Option 2:
[image]

TobKra96, 2021-02-09:
Hi,

the logos look very good. They are elegant and discreet.
My favorites are the ones with the bold font. Spectrum Logo 1 and LED Strip Logo 2.

Some ideas:
- Maybe you can increase the a little bit the saturation. I think you manually decreased it.
- I see the Topics, that the two different logos should address. 
Spectrum -> Music 
LED Strip -> LED Strip / RGB / Technical
I would love it if there is a way to combine two topics inside one logo.
- Maybe we will need an inverted color logo for the sidebar inside the web interface.

Great work. 👍 
Have a nice day!

Teraskull, 2021-02-09:
The colors are from a pastel palette. I can try with a brighter one.

I will experiment with combining them.

I think instead of inverting, a white background would be better.

Teraskull, 2021-02-10:
## Rainbow Spectrum

Light Background:
[image]

Dark Background:
[image]

This logo includes the three RGB colors, the rainbow also represents the Wi-Fi signal, which has a reflection under it, that is an audio spectrum. The logo looks great both on light and dark backgrounds, specifically here is the dark color from the web interface.

In my opinion, it is the best one from all three options.

TobKra96, 2021-02-10:
Good idea.
I am not sure if someone who does not know the idea of the wifi would interpret the rainbow as a wifi signal.
Mhh, If would think about the MLSC, my first thought would be music and led. 

What do you think about these ones:

### Spectrum LED Combined

[image]
[image]

This is the mix between the Spectrum and the LED technical logo.

Teraskull, 2021-02-10:
There are too many LEDs, with a lot of detail, what if you change the LEDs to simple circles, like under the rainbow? I wil try that.

Teraskull, 2021-02-10:
## LED Spectrum

Light Background:
[image]

Dark Background:
[image]

This option is cleaner, and the top "M" is more visible.

In my opinion, this logo, or the Rainbow will be the best looking on the web interface.

The Rainbow logo is less wide, and is possible to put it to the left of "MLSC".

## Rainbow Spectrum Asymmetric

Light Background:
[image]

Dark Background:
[image]

TobKra96, 2021-02-10:
Hi,

my favorite at the moment is your LED Spectrum. Yes, the LEDs in my attempt are too detailed and the "m" is a little bit edgy.
These are the colors of the Webinterface.
```
.blue {
  color: #3498DB; }

.purple {
  color: #9B59B6; }

.green {
  color: #1ABB9C; }

.aero {
  color: #9CC2CB; }

.red {
  color: #E74C3C; }

.dark {
  color: #34495E; }

```
The typical font color is #73879c

Some additional colors:
orange: #F39C12
sky blue: #50C1CF

Could you prepare the LED Spectrum with these colors?
What do you mean?

Have a great day!

Teraskull, 2021-02-10:
Maybe we should also consider the width,like in the rainbow logo? On the top of the web interface, the logo is left to the text, and the LED Spectrum is too long to look good on the left.

What do you think?

Teraskull, 2021-02-10:
## LED Spectrum Left Aligned

Light Background:
[image]

Dark Background:
[image]

As you can see, the width is to big to be placed on the top corner on the web interface.

Also, the purple and red colors do not go well with the dark background.

I think the "M" on the logo looks nice, but it does not match the overall text font.

TobKra96, 2021-02-10:
Yes, the purple does not match. 
And yes it's too big for the horizontal order.

Another idea:
### Volume Meter
[image]

Teraskull, 2021-02-10:
I am checking out your option now

BomBastikDE, 2021-02-10:
LED Strip Logo is nice

Teraskull, 2021-02-10:
I tested the LED Strip Logo and it is also quite detailed, the LED legs are small, and not visible when zoomed out.

I tried flipping the Rainbow upside down, but now it looks like a Hanukkah menorah holder. Your last Rainbow idea is nice, but the spectrum does not make sense when there are multiple colors in the same bar.

I think the Rainbow Spectrum Asymmetric Logo is the cleanest, and looks good when zoomed out. Forget about the Wi-Fi part, it's just the 3 main RGB colors with a reflecting spectrum. It also uses only 3 colors that look good on the dark background, unlike the LED Spectrum, which has 6, and gets detailed.

TobKra96, 2021-02-10:
Ok, understand.

Let's choose the Rainbow Spectrum Asymmetric with horizontal order.
Can we try one last thing? I like the green and red of the logo, but I think the blue-purple does not fit inside the webinterface.

[image]

Can you try to use the typical font color: 73879c
or the blue: 3498DB

Teraskull, 2021-02-10:
Great, your Blue color looks perfect on the dark background!

## Rainbow Spectrum Asymmetric

[image]
[image]

How can I send you the high-resolution PNGs with both separate logo and text?

TobKra96, 2021-02-10:
Perfect!
[image]

Can you zip everything together and upload it here inside the issue? Or you can join the discord server and upload the files there? https://discord.gg/bMmWYGcz
Maybe you can provide 
- the original project file
- a vector file
- two high res png with background 
- one without background?

Great work!
Have a nice day!

Teraskull, 2021-02-10:
Of course I can, one moment, I will tidy up the exports a bit.

Teraskull, 2021-02-10:
Rainbow Spectrum Asymmetric selected as a logo. Issue closed.

closed 2021-02-10 by Teraskull