TobKra96/music_led_strip_control #62: add an wiki to update the system

BomBastikDE, 2021-02-12:
you develop a lot of thinks - could you add an chapter with points we have to do to update the system?

Thanks for your work

Teraskull, 2021-02-12:
Just rerun the automated installation again:

```bash
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash
```

It will automatically save the previous version as a backup, download the new version, and restore your config.

Then, if you use the autostart, restart the service with:

```bash
sudo systemctl restart mlsc.service
```

It should be updated now.

TobKra96, 2021-02-12:
I will add an updated chapter to the wiki tomorrow. 👍

TobKra96, 2021-02-13:
Update Page added to the wiki. [Update Guide](https://github.com/TobKra96/music_led_strip_control/wiki/Update-Guide)
Have a great day!

closed 2021-02-13 by TobKra96