TobKra96/music_led_strip_control #64: Change Configuration File Path

TobKra96, 2021-02-13:
Change the path of the config file to "/share/.mlsc"

Do we need the `music_led_strip_control/server/libs/config.json` any more?
We can create our config if it is missing from the template `music_led_strip_control/server/libs/config.json.bak`.

We have to adjust the import and export mechanism inside the webinterface.
https://github.com/TobKra96/music_led_strip_control/blob/9e67a3b1e90bb417c7778d1667b08f1c437d23a5/server/libs/webserver.py#L26

Teraskull, 2021-02-13:
Nope, I just left it for compatibility right now.

Teraskull, 2021-02-13:
Instead of copying `config.json`:

https://github.com/TobKra96/music_led_strip_control/blob/9e67a3b1e90bb417c7778d1667b08f1c437d23a5/server/libs/config_service.py#L21

We can copy `config.json.bak`.

Anyways, copying the config will ensure that during reinstallation it won't be touched.

TobKra96, 2021-02-13:
Yes, we have to take care of the compatibility of the new config files. If there is a new effect with new settings, the old config file could cause errors.

Additionally, maybe we can use relative paths? Absolute paths are usually no good ideas. What do you mean?

Teraskull, 2021-02-13:
Yes, and a config backup system would be good.

TobKra96, 2021-02-14:
I would split the Topics:

@Teraskull:
- Fix Import/Export Inside web interface (Do we need a backup system too?)
- Change to Relative paths

@TobKra96:
- Create a configuration converter, that will convert old config versions to the newest. Missing fields will be set by the config.json.bak file

What do you think?
Have a great day.

Teraskull, 2021-02-14:
I don't think we need a backup, saving the previous config.json as config.json.bak would be enough.

TobKra96, 2021-02-14:
config.json.bak is the config, we are using for the reset. If everything went wrong, he should able to start from scratch. I would not change it. 
Maybe we can use another name for the user backup. 
Or save the user backup inside the .mlsc folder?

Teraskull, 2021-02-14:
That's what I meant, create a backup with `config.json.bak` in `.mlsc`.

TobKra96, 2021-02-14:
Ok, I was not sure. Yes, that would work.

TobKra96, 2021-02-14:
I added a compatibility checker, which checks every config leaf if it exists and set the default value if it does not exist.
It worked better than I thought.

Teraskull, 2021-02-14:
Perfect, no more manual copying of new effects after updating :D

TobKra96, 2021-02-19:
Done
Maybe we will find a better folder for the config in the future, but for now, it is done.

closed 2021-02-19 by TobKra96