TobKra96/music_led_strip_control #66: Update "Create a custom effect" wiki

Teraskull, 2021-02-13:
Update the wiki to follow the newest changes, for example Font Awesome 5, and new JS imports.

Teraskull, 2021-02-14:
I am currently updating and formatting the wiki here:

```
https://github.com/Teraskull/music_led_strip_control-wiki/wiki/
```

Teraskull, 2021-02-14:
You can update the wiki like this:

```shell
git clone https://github.com/Teraskull/music_led_strip_control-wiki.wiki.git

mv music_led_strip_control-wiki.wiki music_led_strip_control.wiki

cd music_led_strip_control.wiki

git remote remove origin

git remote add origin https://github.com/TobKra96/music_led_strip_control.wiki.git

git push -u origin master
```

Let me know if it works!

TobKra96, 2021-02-14:
Hi, thank you it worked 👍.
I will close this topic. You have already updated the changes, as far as I can see. 
Have a great day!

closed 2021-02-14 by TobKra96