TobKra96/music_led_strip_control #71: Central Logging Library

TobKra96, 2021-02-19:
Check if we can use a logging library, that can save rolling logs, manage the console output.
Convert all print commands to the logging mechanism.

TobKra96, 2021-02-19:
Evaluate https://docs.python.org/3.8/howto/logging.html

Teraskull, 2021-02-20:
Something like this. It has 5 rolling logs up to `5mb` each, and outputs to console.

```python
from logging.handlers import RotatingFileHandler
import logging

logging.basicConfig(handlers=[
                    RotatingFileHandler('mlsc.log', mode='a', maxBytes=5 * 1024 * 1024, backupCount=5, encoding='utf-8'),
                    logging.StreamHandler()
                    ],
                    format='%(asctime)s - %(levelname)s - %(message)s',
                    datefmt='%Y.%m.%d %H:%M:%S',
                    level=logging.INFO
                    )

logger = logging.getLogger('dev')
logger.setLevel(logging.DEBUG)
```

And to disable logging to the console, for example with a checkbox:

```python
logger.StreamHandler(stream=None)
```

closed 2021-03-22 by TobKra96