TobKra96/music_led_strip_control #79: Low output fps, random blinking

d0uble-b, 2021-02-21:
Hi!
I bought a new power supply as you suggested.
And i have issues yet.
I have low fps output and only the first 7-10 LEDs blinking randomly. The service starts with the system
But when i go on mlsc dashboard and select the device and any effect, then nothing happens, all leds are white on the strip. I restarted service and nothing happens.
I send ny config. I hope you can help me.

d0uble_b 

[Config](https://drive.google.com/file/d/1_lVTIx0BAyL-3UWYTThnefQPtB3au7qC/view?usp=drivesdk)

TobKra96, 2021-02-21:
Hi,
what is written on the signal pin of the LED Strip?
Di or Do?

Have a great day!

d0uble-b, 2021-02-21:
Di

TobKra96, 2021-02-21:
Ok, good.

You can try to test the output with this script here:
https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/strandtest.py
Download it and execute it with `sudo python3 strandtest.py`
Ensure MLSC is no running in the background.
If this works, we have to dig deeper into the software. If this does not work, I would take a deeper look at the hardware setup.

Maybe it is a loose connection. Breadboards and stranded wires can cause bad connections.
Do you have other cables, or can you solder them together for a test?

I tested the setup with the same type of poti, you have. It worked for me.
You can drag and drop pictures directly inside the post.
[image]
[image]

d0uble-b, 2021-03-01:
Hello
I think the problem was caused because my raspberry was dying. Unfortunately it happened. I just tried to turn on and its not worked. I tried all tips what i found on topics and forums but i think its hardware error.
So i bought a raspberry pi 4 2gb ram edition and the mlsc worked perfectly like in your video.
Thank you for your help and your time. I am sorry but i didn't know about the raspberry's critical error.

d0uble-b, 2021-03-01:
I will post some pictures about my project when it will be done.

Polar19, 2021-03-04:
Hallo, ich habe das Problem das mein Led Strip auch die ganze zeit blinkt, aber ich weiß nicht woran das liegt. Ich benutze einen Raspberry 4 mit 4gb ram und ws2811 strips

d0uble-b, 2021-03-04:
Use only one usb slot. Use only the microphone. /
Brauchen nur eine usb für mikrofon. Lässt andere usb frei.

Polar19, 2021-03-04:
i use only one usb port for the microphone, but it dont work.

Polar19, 2021-03-04:
the led strip flickers a lot. i tried everything with the potentimeter but i dont have any success with that

d0uble-b, 2021-03-04:
Did you plug the ground cable into the raspberry?

Polar19, 2021-03-04:
yes

d0uble-b, 2021-03-04:
Check the program. Watch the output. My raspberry not always knows my sound card so i have to replug it and restart the program or the pi.

Polar19, 2021-03-04:
my microphone works but the led signal not really

Polar19, 2021-03-04:
i dont get any error message

TobKra96, 2021-03-04:
Hi, just a short summary:
LEDs just flickering, no solid signal.

- Do you use a resistor between the GPIO Pin of the raspberry pi and the LED Strip?
- Have you tested the example script at the top? Does it work? If not, there is a hardware issue.
- Can you share with us some pictures of your setup?

Have a great day!
TobKra

Polar19, 2021-03-04:
1. yes i have a potentiometer between the raspberry and the led strip
2. no i dont tested it with the script, but a day ago it works and it dosent work

Polar19, 2021-03-04:
https://user-images.githubusercontent.com/80054675/110019989-56e65980-7d29-11eb-9d73-d2f218598357.mp4

[image]

TobKra96, 2021-03-04:
Ok, have you tried to adjust the poti value?
Please try the example to differentiate between a hardware issue and a software issue.
https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/strandtest.py

Which gpio pin do you use?
Wich Power Supply do you use? LED Strip and Raspberry Pi share the ground connection?

Polar19, 2021-03-04:
i have connected the raspberry to the ground of the led strip and i use the gpio 18

Polar19, 2021-03-04:
i tried to adjust the poti but it dont work

Polar19, 2021-03-04:
how i can test? just copy and paste the code in to the console?

TobKra96, 2021-03-04:
create a new file with `nano strandtest.py` copy paste the content of the file, adjust the values you need to and execute the script with `sudo python3 strandtest.py`

Polar19, 2021-03-04:
[image]

Polar19, 2021-03-04:
[image]

Polar19, 2021-03-04:
there are two differnt sample rate numbers

Polar19, 2021-03-04:
which is right 44100 or 41000

TobKra96, 2021-03-04:
You have to use the sample rate of your mic.
Do you have discord?
Maybe I can support you with a shared screen.

Polar19, 2021-03-04:
yes

Polar19, 2021-03-04:
lars#2022

TobKra96, 2021-03-04:
Please use the server: https://discord.gg/qfmDu9jnce

TobKra96, 2021-03-06:
Polars issue was fixed by using another power supply.

closed 2021-03-06 by TobKra96

ACyberCat, 2022-02-04:
> Hello I think the problem was caused because my raspberry was dying. Unfortunately it happened. I just tried to turn on and its not worked. I tried all tips what i found on topics and forums but i think its hardware error. So i bought a raspberry pi 4 2gb ram edition and the mlsc worked perfectly like in your video. Thank you for your help and your time. I am sorry but i didn't know about the raspberry's critical error.
> 
> I will post some pictures about my project when it will be done.

Are you really sure it's a hardware error and there's no way around it except for buying a new pi?

My Rpi 3 used to run mlsc perfectly and one day after a clean reinstall it did the same thing, low output fps with random blinking.

Getting my hands on a new pi is near impossible where I currently live due to inflation, so if there is no way around this issue and it really is a hardware fault, i may have to abandon it and keep the lights off.