TobKra96/music_led_strip_control #81: MLSC Server without LAN-Connection

BomBastikDE, 2021-02-25:
actuell I install the MLSC Server as it is explane in the WIKI.
Actuell the MLSC Server starts without LAN connection and I can see the Webside - but the Clients doesn't connect - after Plug in LAN an reboot, everything works well.
Can help someone to confige the system, so I can use it Mobile without LAN?

Best regards

TobKra96, 2021-02-26:
Hi,
I will build the setup as you described and give you feedback tomorrow.
Have a nice day!
TobKra

TobKra96, 2021-02-27:
Hi, I could reproduce the issue.
So I started the Raspberry Pi without Ethernet, only Wifi.
The Wifi Network was not fast enough for the program and the socket was not valid.
Now it will refresh the socket if it is not valid.

Please update your program with 
`curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash`
and send me a short feedback message, if the fix worked for you.

Have a nice day!
TobKra

BomBastikDE, 2021-02-28:
....it works after update!

Thanks all lot - for this Projekt

TobKra96, 2021-02-28:
You are welcome :)

closed 2021-02-28 by TobKra96