TobKra96/music_led_strip_control #82: LED stays off

Jonas1403, 2021-02-26:
Hi all,
really nice project.
I am trying to build the setup with the esp8266 but actually it fails.
The led does not light up.
My suspicion is that the data connection from esp to led cable is broken.
I already measured the voltage that the led cable gets and it fits (round about 5v).
Is it possible to measure the data connection or if data is transmitted?

My esp8266 is use is this:
https://www.azde.ly/8541f9d8a8470612eaa6 (page 16)
I already tried it with the D3 and RX pin, both fails.

I also tried this NeoPixelBus test and it also fails: https://github.com/Makuna/NeoPixelBus/blob/master/examples/NeoPixelTest/NeoPixelTest.ino

Do you have an idea what i am doing wrong?

[image]
[image]
[image]

TobKra96, 2021-02-26:
Hi,

neobus will use spi for the data signal, so we will need the RX pin.
Does the RX pin look like not soldered on the top? Please ensure it is.
As far as I can see, you bridged your fuse?
Have you tried to adjust the Poti?
Where did you measure the 5V? Directly at the LED strip?
Have you checked, you are using the Di pin?

I think it is only for the picture, but you have to be careful running this on your power supply. The housing of the ps is grounded and could cause a short circuit.

Have a nice day.
TobKra

Jonas1403, 2021-02-26:
Hi TobKra,

you are absolutely right. I did not solder the RX pin 🤦🏼‍♂️. 
What a stupid mistake. Now it seems that all works fine.

Thanks for your help and for this fantastic project.

closed 2021-02-26 by Jonas1403