TobKra96/music_led_strip_control #83: Tutorial for Virtual Audio Input/Loopback

TobKra96, 2021-02-27:
Create a Tutorial for a virtual audio input device or a loopback config.
Maybe we can use a Spotify connect app like https://github.com/dtcooper/raspotify

spiro-c, 2021-04-29:
I have done this whit pulseaudio server and [spotifyd](https://github.com/Spotifyd/spotifyd) but you can use any audio source like vlc ... what is playing on the pi 

[image]

then using the [pulsemixer](https://github.com/GeorgeFilipkin/pulsemixer) MLSC can be hook to source you like to get audio from
[image]
[image]

i have use the same method whit the [LedFX](https://github.com/LedFx/LedFx) project but and here is working

TobKra96, 2021-04-30:
Wow, this is really cool. Thank you. I will install it on my system and write a small tutorial 👍

TobKra96, 2021-05-08:
My first test failed. It seems like portaudio and pulseaudio fight against each other 😄.
Pyaudio returned this:

`ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused`

I have already tried to uninstall Portaudio, but PyAudio stopped working after this. 😕 
Don't worry, I will go into the fight for the next round. 😄 

Have a great day
TobKra

spiro-c, 2021-05-08:
I have problem using this on the dev2.2 branch i never manage to get it running but on the master i dont have problem

TobKra96, 2021-05-08:
Ok, thanks for the info. I will check it :)

carlyd95, 2021-07-13:
I was able to get a loopback method working with librespot back in September of last year for my longboard before I broke the damn thing haha. [()](https://carltons-karoo.com/ledboard.mov) And just recently I attempted a different method using bluealsa AD2P. I know the bluealsa AD2P one doesn't require pulse audio or port audio, the librespot one I think required port audio. (tested with RPi 4)

TobKra96, 2021-07-14:
Nice, I will give it a try :)

carlyd95, 2021-07-17:
I got my setup working well and wanted to share... :D

---

Purpose: 
- Alleviate the need for a microphone
- Soley sample the streamed audio (not ambient noise)
- Output audio via USB soundcard

Audio Input Options:
- [BlueALSA](https://github.com/Arkq/bluez-alsa) (Bluetooth A2DP) [confirmed]
- [Librespot](https://github.com/librespot-org/librespot) (Spotify speaker) [confirmed]
- [Shairport Sync](https://github.com/mikebrady/shairport-sync) (Airplay speaker) [possibility]

---

Loopback Tutorial (BlueALSA version):

1. Install BlueALSA
> - `sudo apt install bluealsa`
2. Enabling Audio Profile Sink Role
3. Open the configuration file for bluealsa service
> - `sudo nano /lib/systemd/system/bluealsa.service`
> - Change `ExecStart=/usr/bin/bluealsa` to `ExecStart=/usr/bin/bluealsa -p a2dp-sink`
> - Save changes
4. Reboot
> - `sudo reboot`
5. Start ALSA loop
> - `sudo modprobe snd-aloop`
> - (optional) set up a systemd service or add it in /etc/rc.local that runs this at boot
6. Configure asound.conf
> - `sudo nano /etc/asound.conf`
> - refer to attached asound.conf file (it sets up a multi named both and plays to loopout for MLSC and hw:1,0, the USB soundcard)
7. Connect to A2DP
> - `sudo apt install expect`
> - run attached autobt.sh script file (don't forget file permissions)

OR

> - or use `bluetoothctl`
8.  Direct Audio Stream to "both" previously setup in asound.conf
> - play audio on the device you used to connect to the RPi A2DP
> - `bluealsa-aplay -d both 00:00:00:00:00:00`
> - (optional) set up a systemd service that runs this at boot
9. Start MLSC If Not Running
> - change audio input source to whichever corresponds to loopout (not loopin!)
> - restart MLSC

DONE!

---

Sources Used:
[A2DP tutorial](https://scribles.net/streaming-bluetooth-audio-from-phone-to-raspberry-pi-using-alsa/)

---

Thanks for everyone's work on this project! 

If anyone needs help trying this let me know, and if I made any mistakes or something isn't working just let me know :)

JohnGuan, 2021-08-24:
I used this config from [https://trac.ffmpeg.org/wiki/Capture/ALSA](https://trac.ffmpeg.org/wiki/Capture/ALSA)

**Record audio from an application while also routing the audio to an output device**
```
modprobe snd-aloop pcm_substreams=1
```

```
# /etc/asound.conf
pcm.multi {
    type route;
    slave.pcm {
        type multi;
        slaves.a.pcm "output";
        slaves.b.pcm "loopin";
        slaves.a.channels 2;
        slaves.b.channels 2;
        bindings.0.slave a;
        bindings.0.channel 0;
        bindings.1.slave a;
        bindings.1.channel 1;
        bindings.2.slave b;
        bindings.2.channel 0;
        bindings.3.slave b;
        bindings.3.channel 1;
    }

    ttable.0.0 1;
    ttable.1.1 1;
    ttable.0.2 1;
    ttable.1.3 1;
}

pcm.!default {
	type plug
	slave.pcm "multi"
} 

pcm.output {
	type hw
	card 0 # Your Output Device Name
}

pcm.loopin {
	type plug
	slave.pcm "hw:Loopback,0,0"
}

pcm.loopout {
	type plug
	slave.pcm "hw:Loopback,1,0"
}
```
The mpg123 and shairport-sync are worked.
But MLSC just not working, even it found a list of audio devices.
It drives me crazy, does anyone have work way?

carlyd95, 2021-08-24:
> I used this config from https://trac.ffmpeg.org/wiki/Capture/ALSA
>
> **Record audio from an application while also routing the audio to an output device**
>
> ```
> modprobe snd-aloop pcm_substreams=1 [...]

Does it show the loopout or loopin as one of the sources and you are selecting that source for MLSC to use?

JohnGuan, 2021-08-24:
> Does it show the loopout or loopin as one of the sources and you are selecting that source for MLSC to use?

Ummm... yes.. I tried everyone

carlyd95, 2021-08-24:
@JohnGuan 

in my asound.conf I have my loop configured a lil different, worth a try:

pcm.loopin {
type plug
slave.pcm "plughw:Loopback,0,0"
}

pcm.loopout {
type plug
slave.pcm "plughw:Loopback,1,0"
}

JohnGuan, 2021-08-24:
@carlyd95 Thank you for your generous answer.
In fact, I tried yours. It doesn't work.😂
I think I may need to re-flash the system. My current system is Raspberry Pi OS Lite.
I'm not quite sure if it has any effect, but the PulseAudio solution written in the previous reply seems to be more dependent on the desktop environment.

carlyd95, 2021-08-24:
@JohnGuan weird, mine works just fine using BlueALSA as an A2DP 😆 I’ll have to mess with shairport sometime soon.

Marshalldog, 2022-07-02:
Anyone got any ideas on how turn an MLSC pi into a DLNA speaker?

Marshalldog, 2022-07-02:
or snapcast snapclient?

carlyd95, 2022-07-02:
> or snapcast snapclient?

Snapcast worked when I tried it initially. I dont know if it has support for DLNA though. 

Also bluealsa method isn’t as easy with raspberry pi now as it got removed from the official OS. I haven’t got it to work since.

Marshalldog, 2022-07-07:
what i did
```
sudo apt-get update
sudo apt-get upgrade
sudo reboot
```
I upload snapclient_0.26.0-1_armhf.deb via winscp
ran the following commands
```
sudo apt install vim
sudo dpkg -i snapclient_0.26.0-1_armhf.deb
sudo apt-get -f install
cd /etc/default
sudo vim snapclient
```
inserted the IP address of the host, save and exit

```
sudo update-rc.d snapclient defaults
sudo service snapclient status
```
at this point snapcast was working

then I ran 
```
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b master
sudo reboot
sudo apt-get -f install
```
then when I checked snapcast via:
```
sudo service snapclient status
```
I got:
```
● snapclient.service - Snapcast client
     Loaded: loaded (/lib/systemd/system/snapclient.service; enabled; vendor pr>
     Active: failed (Result: exit-code) since Thu 2022-07-07 08:12:00 BST; 3min>
       Docs: man:snapclient(1)
    Process: 711 ExecStart=/usr/bin/snapclient --logsink=system $SNAPCLIENT_OPT>
   Main PID: 711 (code=exited, status=1/FAILURE)
        CPU: 190ms

Jul 07 08:12:00 raspberrypi systemd[1]: snapclient.service: Scheduled restart j>
Jul 07 08:12:00 raspberrypi systemd[1]: Stopped Snapcast client.
Jul 07 08:12:00 raspberrypi systemd[1]: snapclient.service: Start request repea>
Jul 07 08:12:00 raspberrypi systemd[1]: snapclient.service: Failed with result >
Jul 07 08:12:00 raspberrypi systemd[1]: Failed to start Snapcast client.
```