TobKra96/music_led_strip_control #84: Connection of Rasp with local Wifi instead of Ethernet cable

Boreas83, 2021-03-03:
Hi there
For testing I would like to connect the Rasp to my local Wifi and the ESP8266 too. I would like to make the whole system mobile in the future and access everything with my smartphone. 

I tried to change the arduino sketch but not get it. How to?

Regards
Boreas

TobKra96, 2021-03-03:
Hi,

do you want to use the local network at home or with an own wifi network (Access Point/Hotspot)?
What is the issue? Can you describe the problem?

Have a nice day!
TobKra

BramvdHoek, 2021-03-05:
Hello, I have the same question:

I want to use this system in a place where I can't put down an Ethernet cable to connect it wired to my home network.
Instead, I would connect my Raspberry Pi to my home network wireless.
And control (to start off) one led strip:

--------------     (wireless)       --------- 
{Acces point}   < - - - - - - >    | Raspi4|--- > LED strip 1 
--------------                            ---------
          | 
          | 
    ---------
    | Phone|
    ---------
Is this possible? I want eventually to go to a situation where I can control multiple LED strips directly from the raspberry pi by changing in the settings the pin layout for each separate LED strip.

--------------     (wireless)       ---------          -- -- -- > LED strip 1 
{Acces point}   < - - - - - - >    | Raspi4|- - < |
--------------                            ---------           -- -- -- > LED strip 2
          | 
          | 
    ---------
    | Phone|
    ---------

In short: When the raspberry pi 4 isn't connected wirelessly, am I able to connect to it to control the LED strips.

TobKra96, 2021-03-06:
Hi,

Yes, the 1. will work.

In theory, the output library of the LEDs on the raspberry pi supports 2 stripes: https://github.com/jgarff/rpi_ws281x
But I did not test it. So in theory, yes.

Have a great day!
TobKra

closed 2021-03-22 by TobKra96