TobKra96/music_led_strip_control #87: Make use of the SK6812 RGBW white channel?

TimoS911, 2021-03-15:
Is it somehow possible to make use of the white channel on SK6812 LED Strips?
I can only see coding for RGB colors and not the W.

In current rpi_ws281x git there is a strandtest.py which also adresses the white channel on its own.
But if i run the script to get only warm white ambient light and want to swap back to music controlled, i have to reboot the pi because MLSC is "crashed" or at least doesnt switch effects anymore.

Would be nice to have some option to write own effects which use the white channel. But this seems not doable with the current guide for writing own effects.

TobKra96, 2021-03-16:
Hi,
at the moment there is no implementation for the white channel.
The efforts are very high to implement it in every effect. I do not see the big user base, that will have a benefit of the white channel.

If MLSC is running and you start the strandtest, the strandtest will claim, the gpio pin by initializing the library.
You can either press the save button inside the general settings page or restart the mlsc service to reinit the gpio pin for mlsc again.

If you want to create your own effect with a white channel, you have to adjust the array, which will be put into a queue for communication between the effect service and the output service. Maybe you get some trouble with other existing effects, that do not know the new array format.

An easy implementation, but maybe not as functional as you want to:
You can check and replace the rgb value with one white value at the output.
If you see a led that get the values r:255 g:255 b:255 you can replace it with r:0 g:0 b:0 w:255
r:255 g:255 b:255 --> r:0 g:0 b:0 w:255
r:100 g:100 b:100 --> r:0 g:0 b:0 w:100

You have to adjust the logic here:
https://github.com/TobKra96/music_led_strip_control/blob/453ef9af70e9d177efc36d57ff5d5ecc7adc3f52/server/libs/outputs/output_raspi.py#L87

Have a great day!
TobKra

TimoS911, 2021-03-23:
Okay i am to stupid for that implementation...
I will take a look in the tutorial for a new effect to get whats going on in python... will take some time i think.

I get that it's to much effort to implement the white channel in everything. And it's not really necessary as many effects look way better with 255,255,255 white then 0,0,0,255 warm white.
But maybe implement the white channel in the single effect?
That would be everything i need :)
Or is the whole thing not knowing what to do with a 4th variable/array so that changing 1 effect is like changing everything?

Have a good one,
Timo

TobKra96, 2021-07-11:
done with 2.2

closed 2021-07-11 by TobKra96