TobKra96/music_led_strip_control #92: No Connect after Update

BomBastikDE, 2021-03-23:
I done an Update, as it is discribed in Wiki
- No connect via http://
- not on Port 80 - not on Port 8080

BomBastikDE, 2021-03-23:
sudo python3 main.py 
Manual start doesn't work:
root@raspberrypi:/share/music_led_strip_control/server# sudo python3 main.py
Traceback (most recent call last):
  File "main.py", line 13, in <module>
    from libs.audio_process_service import AudioProcessService
  File "/share/music_led_strip_control/server/libs/audio_process_service.py", line 3, in <module>
    from libs.config_service import ConfigService  # pylint: disable=E0611, E0401
  File "/share/music_led_strip_control/server/libs/config_service.py", line 9, in <module>
    import coloredlogs
ModuleNotFoundError: No module named 'coloredlogs'

BramvdHoek, 2021-03-23:
Would like to hang on this issue too:

After installing the automatic script, we need to configure the Led strip.
According to: Configure Music LED Strip Control

At this stage, MLSC and all dependencies should be installed. Now we can execute the program and configure the main settings. Navigate to the server directory:

`cd /share/music_led_strip_control/server`
Run the program manually:

`sudo python3 main.py`

Eventually the error that BomBastikDE stated is rolling out:
`
pi@raspberrypi:/share/music_led_strip_control/server $ sudo python3 main.py 
Traceback (most recent call last):
  File "main.py", line 13, in <module>
    from libs.audio_process_service import AudioProcessService
  File "/share/music_led_strip_control/server/libs/audio_process_service.py", line 3, in <module>
    from libs.config_service import ConfigService  # pylint: disable=E0611, E0401
  File "/share/music_led_strip_control/server/libs/config_service.py", line 9, in <module>
    import coloredlogs
ModuleNotFoundError: No module named 'coloredlogs'`

Teraskull, 2021-03-23:
@BomBastikDE @BramvdHoek, sorry guys, I forgot to include the new modules in the installation script. For now you can run the following inside `/share/music_led_strip_control/`:

```bash
sudo pip install -r requirements.txt
```

BomBastikDE, 2021-03-23:
For now you can run the following inside `/share/music_led_strip_control/`:
> 
> ```shell
> sudo pip3 install -r requirements.txt
> ```

Workaround runs - Thx

closed 2021-03-23 by BomBastikDE