TobKra96/music_led_strip_control #94: is it possible to trigger Effects from extern

BomBastikDE, 2021-03-23:
Is it posible to trigger the effects from Extern (like an Smarthome Tablet) over an Http-adress?
I want insert the controll of MLSC in my ioBroker Project.

TobKra96, 2021-03-23:
Hi,
we use a Rest API to communicate between the web interface.
Maybe you can use an adapter for this.

E.g. Rest enpoint: http://<raspberry-pi-ip>/SetActiveEffectForAll
Json content: {effect: "effect_gradient"}

See this issue: https://github.com/TobKra96/music_led_strip_control/issues/48
PHP Example: https://gist.github.com/TobKra96/7004030cfb846f9792b3ec2afe423275
Available effects: https://github.com/TobKra96/music_led_strip_control/blob/master/server/libs/effects_enum.py

Teraskull, 2021-04-07:
Adding some full examples, so that others can find this answer. This works for both Linux and Windows:
```shell
curl -X POST -H "Content-Type: application/json" -d "{\"effect\": \"effect_gradient\"}" http://[YOUR_IP]:8080/SetActiveEffectForAll
```

```python
import requests

r = requests.post('http://[YOUR_IP]:8080/SetActiveEffectForAll', json={"effect": "effect_gradient"})
```

TobKra96, 2021-07-11:
done, I will add a entry inside the faq

closed 2021-07-11 by TobKra96