TobKra96/music_led_strip_control #102: random mode

dannysprogis, 2021-04-08:
What an amazing setup! 

I am learning raspberry pi and love this project! I am currently trying to write my own effects but thought I could share my ideas so people can actually make it work while i fumble around trying...

it would be really cool in general to add some random options to most of the existing effects so they become a little more dynamic. for example the direction changer effect would look better if you could set it to fire off pulses a random number of times in each direction. so the controls could be random enable switch and an upper and lower threshold for the random generator to choose values between. 

A second feature would be really cool to have a cycle effect that randomly selects effects and cycles through them for a set amount of time. Perhaps you could choose via checkboxes the effects that you would like to include? configurations could be time between change, random choose effect, cycle effect from check list. 

for now these are beyond my capability but that's my goal. any

Teraskull, 2021-04-08:
Hi, thanks!

The second feature is currently on our list, and yes, the config will include the time between change and checkboxes.

Regrading the first feature, unfortunately, I don't think it will be easy to change the animation logic to add randomness for most effects, but we will see what we can do.

dannysprogis, 2021-04-09:
Hi, thanks for the quick response! that's awesome news to hear. Is there a timeline for the cycle effect functionality to be added and where should I look to see the updates? 

I wish I could be more helpful with the coding but I have basically no experience with Linux and web interfaces but am willing to learn and am trying...

Teraskull, 2021-04-09:
Sorry, no exact timeline, but there will be a big update in around 2-3 months, which will include this feature.

TobKra96, 2021-07-11:
done with 2.2

closed 2021-07-11 by TobKra96