TobKra96/music_led_strip_control #106: Add Troubleshooting / FAQ

TobKra96, 2021-04-12:
Add the most typical issues and document it.

BramvdHoek, 2021-04-13:
Which potmeter do you use? The one I use doesn't really work

TobKra96, 2021-04-13:
I am using a 1k potentiometer. 
You can share your setup and settings, so we can help you with the issue you have.

BramvdHoek, 2021-04-20:
Led strip:  WS2815 
Length: 5m (might cutt-off a bit)
No led per m: 60 (300 total)

Potmeter:

|  Ω | Solid color on|LED off|
| ------------- | ------------- |------------- |
| 16  | Seems ok, sometimes a little flicker at a few led's | quite a few leds flicker|
| 100  | LED solid on (like expected) | Even more leds flicker, seems random |
| 150  | LED on, all diff colors. Part stays on while the leds on the end are off. |Leds are on, last ones have random color. First one are solid blue|

TobKra96, 2021-04-20:
Check your ground connection (shared ground between led strip, raspi and power supply), use short cables to increase the signal quality. Maybe solder the cables instead of just stick them into a breadboard (increase signal quality too).

Please see https://github.com/TobKra96/music_led_strip_control/issues/111 . He had flickering issues, caused by ground issues.

**Can you try this:**
#### PWM

Since this library and the onboard Raspberry Pi audio
both use the PWM, they cannot be used together.  You will need to
blacklist the Broadcom audio kernel module by creating a file
`/etc/modprobe.d/snd-blacklist.conf` with

    blacklist snd_bcm2835

If the audio device is still loading after blacklisting, you may also
need to comment it out in the /etc/modules file.

On headless systems you may also need to force audio through hdmi
Edit config.txt and add:

    hdmi_force_hotplug=1
    hdmi_force_edid_audio=1

A reboot is required for this change to take effect

Some distributions use audio by default, even if nothing is being played.
If audio is needed, you can use a USB audio device instead.

[original link](https://github.com/jgarff/rpi_ws281x/blob/master/README.md)

BramvdHoek, 2021-04-20:
> Check your ground connection (shared ground between led strip, raspi and power supply), use short cables to increase the signal quality. Maybe solder the cables instead of just stick them into a [...]

Thanks for your reply.
Before I tried the above, I did change the **LED Signal Frequenzy in Hz**
 setting to 850000. Somehow it did stop flickering. Even without potmeter.

TobKra96, 2021-04-21:
Mhh? WS2815 uses 800kbit. Strange that it works with 850 kbit.

TobKra96, 2021-11-23:
FAQ Added

closed 2021-11-23 by TobKra96