TobKra96/music_led_strip_control #110: instruction issue

Gutleben, 2021-04-17:
Hi!
First, I want to thank you very much for this excellent project. I've been looking for so long for a nice led color solution based on light. After i took all my motivation together and decided to write it on my own, i found your project.
It works nice.

I want to participate on the project and create more custom effects.

About the issue:
In the instructions the path to the template for custom effects written:
/share/music_led_strip_control/server/libs/app/home/templates/dashboard.html 
the current correct path is:
/share/music_led_strip_control/server/libs/templates/dashboard.html 

Thanks a lot for your work!

Gutleben, 2021-04-17:
Also the add of custom effects in the dashboard is not accurate atm
<li id="effect_custom">
    <i class="fa fa-check dasboard-glyph" aria-hidden="true"></i>
    <span class="glyphicon-class">Ivos Effekt</span>
</li>

TobKra96, 2021-04-17:
Hi,
you are right, we have already updated the paths and the instructions for the upcoming update.
Sorry for the confusion.
You can take a look at an older revision of the guide:
https://github.com/TobKra96/music_led_strip_control/wiki/Create-a-custom-effect/141b8d594400748282f9277a49344f62047c52a9

Have a great day!
TobKra

Gutleben, 2021-04-17:
Thanks, that was no problem for me, to get through this!

closed 2021-07-11 by TobKra96