TobKra96/music_led_strip_control #111: How to fix WS2811 12V LED strip flickering

dotnetaddicted, 2021-04-17:
Everything is setup like at the image - [link](https://user-images.githubusercontent.com/7833146/105534965-88adde80-5cee-11eb-980e-525cf88ca1da.jpg), but I'm unable to get stable strip color by adjusting B2K potentiometer 2K Ohm. 
For example "Single Non-Music effect" should be green, but it flickers with green, red and blue colors as shown [here](https://drive.google.com/file/d/1xRqC5Bhdxcx-ovP9MCuDnswnPBZ6e2DQ/view?usp=sharing), but actually any effect has this flickering.

- LED strip - WS2811 12V 60 LED/m, length of strip is 2.7m.
- Power supply - LP-12120 (12V/10A).
- Raspberry PI 2 Model B (powered from USB hub with 5V/4A power supply)

P.S.: If to connect WS2812B 5V LED strip I can easily adjust it with the potentiometer and it's not flickering, all effects work just fine. Of course 12V power supply is not involved in this scheme, so I believe the problem is caused by some grounding issue.

Would you suggest something?
Thanks,

TobKra96, 2021-04-17:
Hi,

do your WS2811 strip share ground with the Power Supply and the Raspberry Pi?
Can you share your device config and maybe some pictures of your setup?

**Can you try this:**
#### PWM

Since this library and the onboard Raspberry Pi audio
both use the PWM, they cannot be used together.  You will need to
blacklist the Broadcom audio kernel module by creating a file
`/etc/modprobe.d/snd-blacklist.conf` with

    blacklist snd_bcm2835

If the audio device is still loading after blacklisting, you may also
need to comment it out in the /etc/modules file.

On headless systems you may also need to force audio through hdmi
Edit config.txt and add:

    hdmi_force_hotplug=1
    hdmi_force_edid_audio=1

A reboot is required for this change to take effect

Some distributions use audio by default, even if nothing is being played.
If audio is needed, you can use a USB audio device instead.

[original link](https://github.com/jgarff/rpi_ws281x/blob/master/README.md)

dotnetaddicted, 2021-04-18:
Hi @TobKra96,

_"do your WS2811 strip share ground with the Power Supply and the Raspberry Pi?"_
Yes, sure.

I've checked everything once again and there is no flickering when I use shorter connectors and a breadboard.
But when I use longer cables for ground the issue is here :(

TobKra96, 2021-04-18:
Hey, 

ok, so the issue was caused by a too-long ground cable, which didn't proper ground all components.
I will add this experience inside the panned FAQ.
Let me know if you have further issues 😃 .

Have a great day
TobKra

dotnetaddicted, 2021-04-19:
Thanks for your help, I'm closing the issue.

closed 2021-04-19 by dotnetaddicted