TobKra96/music_led_strip_control #114: System Info page

TobKra96, 2021-04-18:
Add a page that polls data like:

- CPU
- RAM
- Network usage
- Temperatures
- Running Services
- hostapd service status

Maybe you can restart hostapd.

Teraskull, 2021-04-18:
For CPU, RAM and Network Usage we can use [psutil](https://pypi.org/project/psutil/).

For Temperature we should use the dedicated software for raspberry pi:
```shell
/opt/vc/bin/vcgencmd measure_temp
```
Which in Python can be used like this:
```python
import os
import time

def measure_temp():
    temp = os.popen("vcgencmd measure_temp").readline()
    return (temp.replace("temp=",""))

print(measure_temp())
```

Getting running services and hostapd status can be done with `os` or `subprocess`.

Teraskull, 2021-04-19:
Here is a good Flask example: https://github.com/Jahaja/psdash

TobKra96, 2021-04-23:
API Implementation started: https://github.com/TobKra96/music_led_strip_control/commit/95557c85e42b3f8f75846906cfea4b6503131ac5 😎 

Added GetSystemInfoPerformance, GetSystemInfoTemperature, GetSystemInfoServices Endpoints

[image]

TobKra96, 2021-07-11:
done with 2.2

closed 2021-07-11 by TobKra96