TobKra96/music_led_strip_control #115: Some effects not working

mingersming, 2021-04-20:
Hey,

I'm very lucky that I found this masterpiece of a gem. Everything is working flawlessly right from the start, except 3 effects that are not working. 

+ Beat
+ Beat Twinkle
+ Direction Changer

My specs:
+ Pi 3 B+
+ 2xLED stripes (WS2812E, 2x5m, DC5V)
+ Icy Box AC527 USB External Sound Card

Any ideas? 

Best regards

Teraskull, 2021-04-20:
Hi, these effects react to bass, so you will not see the lights workif you, for example, clap. Make sure your microphone picks up the base range.

Test with a different USB microphone, and see if it changes anything.

You can also try changing the microphone sensitivity in the settings, you can find instructions in the [wiki](https://github.com/TobKra96/music_led_strip_control/wiki/Adjust-microphone-volume-level).

Thanks!

TobKra96, 2021-07-11:
done

closed 2021-07-11 by TobKra96