TobKra96/music_led_strip_control #117: ESP8266 serial monitor issue

Rotondo27, 2021-04-25:
Thank you for your excellent work. The raspberry pi part of the project is working great!
I am having an issue with the esp client and would really appreciate your help. When I flash to the esp using arduino ide, everything looks how it does in your installation guide. When I open the serial monitor, I just received a bunch of question marks (as shown in the picture) instead of the client actually running the software.
Of note, the blink example program DOES work when I flash it to the client. 
I bought the esp from Amazon in the following link:
https://www.amazon.com/dp/B081CSJV2V/ref=dp_iou_view_product_mobile?ie=UTF8&psc=1
The manufacturer settings are shown in the screenshot from their website (attached below).
Here are the steps I took to flash the software:
1. Downloaded arduino IDE
2. Copied board manager url to preferences
3. Downloaded esp2866 community board manager
4. Downloaded "Wifi" and "neopixelbus" libraries
5. Copied client code from github exactly and made changes to LED number, ssid, password, and ip addresses
6. Flashed firmware (compiled and flashed with confirmation messages in IDE that matched the installation guide)
7. Opened serial monitor, which showed mostly backwards question marks and non sense despite resetting. The baud rate on the serial monitor is 9600, which matches the stated value on the back of the esp.

I did not download the nodemcu flasher, maybe that could be the issue? But I thought that arduino ide overwrites this firmware anyway, so it would not be needed

Board settings:
[image]

Sketch
[image]

Serial monitor
[image]

Manufacturer recommended settings for esp device
[image]

spiro-c, 2021-04-26:
change baud rate to 115200 serial monitor is on that rate you can see it in the sketch
    https://github.com/TobKra96/music_led_strip_control/blob/33d6e9ca3c5cc699f1f700e6bf00727fa2f65f3c/client/udp/esp/nodemcu_client/nodemcu_client.ino#L34-L36

TobKra96, 2021-11-23:
baud rate fixed it

closed 2021-11-23 by TobKra96