TobKra96/music_led_strip_control #122: MLSC is working whit WLED

spiro-c, 2021-04-29:
I just test this option and i can confirm  it is working and there is no need to setup access point on the pi you can use existing wifi network and up to 490 pixels can be driven over UDP
For me is working if i add [WLED](https://github.com/Aircoookie/WLED/wiki/UDP-Realtime-Control) like UDP device on port 19446 what is port for Hyperion but MLSC is sending same data 

[image]

Teraskull, 2021-04-29:
Thank you for this insight, we will add it to the wiki.

TobKra96, 2021-04-30:
Cool, thank you for the testing.

carlyd95, 2021-07-13:
You can also add WS2813 tested with RPi 4, RPi 4B and ESP8266 NodeMCU (on longer strips the first pixel will have issues if one chooses not to use a logic level shifter though...)

TobKra96, 2021-07-14:
Thank you. I added it to the [Hardware Compatible List](https://github.com/TobKra96/music_led_strip_control/wiki/Compatible-Hardware-List) .

closed 2021-07-14 by TobKra96

sergio-faraldo, 2021-12-09:
@carlyd95 I am trying to set up mlsc with a WS2813 strip. How did you set it up? I'm asking because all the diagrams on this github are for 3 pin LEDs, and i don't know what to do with the 4 pins the WS2813 strip has

carlyd95, 2021-12-09:
> @carlyd95 I am trying to set up mlsc with a WS2813 strip. How did you set it up? I'm asking because all the diagrams on this github are for 3 pin LEDs, and i don't know what to do with the 4 pins the WS2813 strip has

Two middle are both data lines, one being a backup; I connected them together and connected the signal wire to both. 👍🏽