TobKra96/music_led_strip_control #123: ESP Client

Gutleben, 2021-05-02:
Is it possible to use an esp client in an existing network?
I have a place where the mlsc runs where no ethernet connection is possible. And i can't see any limitions to work on an existing wifi.

Thanks for the answer

Gutleben, 2021-05-02:
maybe a duplicate of https://github.com/TobKra96/music_led_strip_control/issues/122

Gutleben, 2021-05-02:
Ok, i can confirm that esp32 is working with udp over an already existing wifi.
Like described in the wiki: [Here](https://github.com/TobKra96/music_led_strip_control/wiki/Installation-Guide#setup-esp8266)
We have to configure the esp32.
You have to add these libaries: 
`#include <WiFi.h>` instead of the ESP8266WiFi libary.

Than flash and start the program.

After that go to mlsc, add a new device with the ip-address of the device and the port (default 7777)
works like a charm!

Love this project!

TobKra96, 2021-05-02:
Hi, yes this will work but
- decrease the performance/stability. The package uses a longer way.
- increase the network traffic (spam) of your home network. This will decrease the speed of other wifi devices in the network.

If you are fine with these limitations, you can use your home network.

Have a great day
TobKra

closed 2021-07-10 by TobKra96