TobKra96/music_led_strip_control #124: VC4CL and ClPy support for Raspberry Zero OpenCL GPU speedup

jumd, 2021-05-03:
Hello,

maybe it is possible to run MLSC on a Raspberry Zero and save some power or be able to run it moblie with a power bank.

VC4CL allows to use OpenCL on Raspberry 1 to 3 GPUs, but not Rasperry 4: https://github.com/doe300/VC4CL
ClPy can execute CuPy Code on OpenCL devices: https://github.com/fixstars/clpy

Do you think this can speed up the computations of numpy and scypy to be able to run on a Raspberry Zero?
Maybe there could be already much speed improvement by only doing significant parts like FFT and else in the GPU.

This would be really interesting for me, but im not sure if it really could work and it will take much time for me to get into this and try it by myself as im not much into python and its libraries.

Maybe it will be more easy for me to buy a Banana Pi M2 Zero which should be fast enough to run MLSC and able to be supplied by a power bank.
I think using the GPU of Raspberry Zero could be more power efficient than using Banana Pi.

Thank you for your opinion on this and thanks for this cool Software.

Regards
jumd

TobKra96, 2021-05-08:
Hello jumd,

thank you for the idea with VC4CL.
I don't think, we will implement the gpu acceleration. To be honest, I won't spend so many hours in performance optimization for low-tier hardware to gain maybe +10 fps.
Additionally, I want to support the newer Raspberry Pi platforms, like the Pi 4. Therefore I would need to maintain two different software components, that will process the music array. 😞 
MLSC is multicore optimized and the focus is on multiple device support.
We see more potential inside the Web interface optimization and more user-friendly installation.

Sorry for the potential disappointment, but I want to be transparent with you. 👍 

Have a great day
TobKra

jumd, 2021-05-08:
Yes, I totally understand it. I already bought a Banana Pi Zero and will try running MLSC on it.
I can report back to you how it works. It should be fine with Quad Cortex A7 at 1,2Ghz.

Thank you for your kind answer and the work you put in this amazing project.
jumd

TobKra96, 2021-07-11:
Not implemented

closed 2021-07-11 by TobKra96