TobKra96/music_led_strip_control #126: Is it possible to change effects without use of web interface?

beardfacefacebeard, 2021-06-05:
i have everything running and it all works quite well. the effects are amazing.. but when i implement this into the project i am working on the rpi will not have internet access at all and will operate headless. 
im not really much for coding but i can do some simple things.. i was hoping to be able to cycle through effects with a tactile button press or a rotary switch if possible. im not sure if this can be easily attained or not, and if not that is ok. 
thanks for your time, and i appreciate all the work that has already been put into this. it is truly fantastic.

Teraskull, 2021-06-05:
Hi, yes, #94 is the correct way to access the API to control lights without the interface. We recently changed the API endpoint names in branch `dev_2.2`, but the idea is the same.

beardfacefacebeard, 2021-06-06:
thanks for the response. follow up question, if i have multiple led devices (ie MainLED, SecondLED) and only want to address the change in one how would i vary that?
`SetActiveEffectForAll` is obviously going to change them all, would i use `SetActiveEffectForMainLED` -- using python

Teraskull, 2021-06-06:
@beardfacefacebeard Check out our `dev_2.2` branch, there is a complete API documentation with examples. To be able to select a specific device, you should pass the device id as a query, like here:

[image]

Also I strongly suggest to build your scripts with the new API endpoints, since the ones in `main` branch will not be backwards compatible after the new release.

beardfacefacebeard, 2021-06-06:
im really sorry for my ignorance but i have no idea where that is. im fairly new to github and cant find the documentation youre referring to. i feel completely lost. i appreciate your patience

Teraskull, 2021-06-06:
@beardfacefacebeard In the branch selector, choose `dev_2.2` and install that version.

[image]

beardfacefacebeard, 2021-06-07:
oh, no wonder i was so lost.. thank you. this really is some amazing stuff. i appreciate it all so much.

closed 2021-06-07 by beardfacefacebeard