TobKra96/music_led_strip_control #128: Flickering

llamatude, 2021-06-19:
I am using https://github.com/jgarff/rpi_ws281x for another project and it works well no issues. For some reason with MLSC i am getting flickering when on solid colors. I assume its doing it on the music as well but so much is going on i can tell. Everything else seems to work well just this odd flicker.

Teraskull, 2021-06-21:
Hi,

Can you write the steps to reproduce the issue?

How many devices are created and does the flickering continue if only one device is active?

Also, are you using the `master` or `dev_2.2` branch?

llamatude, 2021-06-21:
If i turn it on solid color or anything really it just flickers. 

I am running raspberry pi no remote modules yet

i am running master.. From what i can tell is when it updates the colors is when i get the flickering.

Teraskull, 2021-06-21:
Could you please try running https://github.com/TobKra96/music_led_strip_control/tree/dev_2.2 and tell if the issue persists?

No other setup is required except for cloning/pulling the repo, switching the branch to `dev_2.2`, and installing pip modules with:
```
sudo pip3 install -r requirements
```

llamatude, 2021-06-21:
still having the issue.. I made a small video for you to see:

https://youtu.be/BcDKlZWVmls

the code i am using that works no flicker is this:
`def Blank(strip):
    SetAll(strip, Color(255, 0, 0))
    strip.show()
`
I feel like its a power issue I have 2 strips at 54 lights per strip. I set the lights to 54 but I think its trying to send the signal to 124 lights. I do have a SPI Signal Amplifier Repeater. 
https://www.amazon.com/gp/product/B07LGM1694

llamatude, 2021-06-21:
NVM this is a me issue... i just did a test with and i am getting the same flicker.. odd now time to trace this down! sorry for the false alarm. 

    for i in range(0, 3):
        #Fade In.
        for j in range (0, 256):
            if i == 0:
                SetAll(strip, Color(j, 0, 0))
            elif i == 1:
                SetAll(strip, Color(0, j, 0))
            elif i == 2:
                SetAll(strip, Color(0, 0, j))
            strip.show()
        #Fade Out.
        for j in range (256, 0, -1):
            if i == 0:
                SetAll(strip, Color(j, 0, 0))
            elif i == 1:
                SetAll(strip, Color(0, j, 0))
            elif i == 2:
                SetAll(strip, Color(0, 0, j))
            strip.show()

Teraskull, 2021-06-21:
No problem, thanks for testing and confirming!

TobKra96, 2021-07-10:
I think the issue can be closed :)

closed 2021-07-10 by TobKra96