TobKra96/music_led_strip_control #131: coloredlogs module

Wraith99, 2021-07-15:
Hi, sorry ive seen your responses to the coloredlogs module missing error but mine hasn't been fixed by the requirements.txt :

   __pi@raspberrypi:/share/music_led_strip_control/server $ sudo python3 main.py
Traceback (most recent call last):
  File "main.py", line 13, in <module>
    from libs.audio_process_service import AudioProcessService
  File "/share/music_led_strip_control/server/libs/audio_process_service.py", line 3, in <module>
    from libs.config_service import ConfigService  # pylint: disable=E0611, E0401
  File "/share/music_led_strip_control/server/libs/config_service.py", line 10, in <module>
    import coloredlogs
ModuleNotFoundError: No module named 'coloredlogs'
pi@raspberrypi:/share/music_led_strip_control/server $__

trying the requirements =.txt yields the following :
  
pi@raspberrypi:/share/music_led_strip_control $ ​sudo pip3 install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: PyAudio==0.2.11 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 1)) (0.2.11)
Requirement already satisfied: numpy==1.20.3 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 2)) (1.20.3)
Collecting coloredlogs==15.0
  Using cached https://www.piwheels.org/simple/coloredlogs/coloredlogs-15.0-py2.py3-none-any.whl (45 kB)
Requirement already satisfied: scipy==1.6.2 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 4)) (1.6.2)
Requirement already satisfied: Flask==2.0.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 5)) (2.0.1)
Collecting Flask_Login==0.5.0
  Using cached https://www.piwheels.org/simple/flask-login/Flask_Login-0.5.0-py2.py3-none-any.whl (16 kB)
Requirement already satisfied: Jinja2==3.0.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 7)) (3.0.1)
Requirement already satisfied: waitress==2.0.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 8)) (2.0.0)
Requirement already satisfied: rpi_ws281x==4.3.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 9)) (4.3.0)
Collecting icmplib==3.0
  Using cached https://www.piwheels.org/simple/icmplib/icmplib-3.0-py3-none-any.whl (30 kB)
Collecting flasgger==0.9.5
  Using cached https://www.piwheels.org/simple/flasgger/flasgger-0.9.5-py2.py3-none-any.whl (3.9 MB)
Collecting psutil==5.8.0
  Using cached https://www.piwheels.org/simple/psutil/psutil-5.8.0-cp37-cp37m-linux_armv7l.whl (296 kB)
Requirement already satisfied: humanfriendly>=9.1 in /usr/local/lib/python3.7/dist-packages (from coloredlogs==15.0->-r requirements.txt (line 3)) (9.2)
Requirement already satisfied: itsdangerous>=2.0 in /usr/local/lib/python3.7/dist-packages (from Flask==2.0.1->-r requirements.txt (line 5)) (2.0.1)
Requirement already satisfied: Werkzeug>=2.0 in /usr/local/lib/python3.7/dist-packages (from Flask==2.0.1->-r requirements.txt (line 5)) (2.0.1)
Requirement already satisfied: click>=7.1.2 in /usr/local/lib/python3.7/dist-packages (from Flask==2.0.1->-r requirements.txt (line 5)) (8.0.1)
Requirement already satisfied: MarkupSafe>=2.0 in /usr/local/lib/python3.7/dist-packages (from Jinja2==3.0.1->-r requirements.txt (line 7)) (2.0.1)
Requirement already satisfied: PyYAML>=3.0 in /usr/local/lib/python3.7/dist-packages (from flasgger==0.9.5->-r requirements.txt (line 11)) (5.4.1)
Requirement already satisfied: jsonschema>=3.0.1 in /usr/local/lib/python3.7/dist-packages (from flasgger==0.9.5->-r requirements.txt (line 11)) (3.2.0)
Requirement already satisfied: mistune in /usr/local/lib/python3.7/dist-packages (from flasgger==0.9.5->-r requirements.txt (line 11)) (0.8.4)
Requirement already satisfied: six>=1.10.0 in /usr/lib/python3/dist-packages (from flasgger==0.9.5->-r requirements.txt (line 11)) (1.12.0)
Requirement already satisfied: importlib-metadata in /usr/local/lib/python3.7/dist-packages (from click>=7.1.2->Flask==2.0.1->-r requirements.txt (line 5)) (4.6.1)
Requirement already satisfied: attrs>=17.4.0 in /usr/local/lib/python3.7/dist-packages (from jsonschema>=3.0.1->flasgger==0.9.5->-r requirements.txt (line 11)) (21.2.0)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from jsonschema>=3.0.1->flasgger==0.9.5->-r requirements.txt (line 11)) (40.8.0)
Requirement already satisfied: pyrsistent>=0.14.0 in /usr/local/lib/python3.7/dist-packages (from jsonschema>=3.0.1->flasgger==0.9.5->-r requirements.txt (line 11)) (0.18.0)
Requirement already satisfied: zipp>=0.5 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata->click>=7.1.2->Flask==2.0.1->-r requirements.txt (line 5)) (3.5.0)
Requirement already satisfied: typing-extensions>=3.6.4 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata->click>=7.1.2->Flask==2.0.1->-r requirements.txt (line 5)) (3.10.0.0)
Installing collected packages: psutil, icmplib, Flask-Login, flasgger, coloredlogs
  Attempting uninstall: psutil
    Found existing installation: psutil 5.5.1
ERROR: Cannot uninstall 'psutil'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
pi@raspberrypi:/share/music_led_strip_control $

I have a feeling its a path error with the installation in my case but I haven't been able to figure out how to fix it , any advice would be super appreciated. Thanks

TobKra96, 2021-07-17:
Please use the newest update of mlsc. You can install it via:

`curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b master
`

This should install the missing lib.

Wraith99, 2021-07-21:
Hey @TobKra96 ,  so i reinstalled it as you recommended getting some new issues now. After running main.py I get the following output;

pi@raspberrypi:/share/music_led_strip_control/server $ sudo python3 main.py
INFO     - __main__                       - Initializing MLSC...
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM dsnoop:1
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM dmix:1
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 1}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card

Teraskull, 2021-07-21:
Does MLSC still work with these errors?

Wraith99, 2021-07-22:
it runs but the pi seems to have no interaction with the strip. After several hours adjusting parameters I'm starting to wonder whether its an issue with my wiring. As I'm using WS2812b strip I was wondering if I require a logic level shifter between the pi and the strip?

Wraith99, 2021-07-22:
Alright, further update. Got the strip to run after following the instructions on: https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/ strip ran fine. I then changed back to MLSC and re-ran it and started non-music effects flawlessly.... my guess is a missing dependency that was accidentally fixed when I ran the above test installation. 

The music-related effects are proving to be a problem however... I'm still getting the following output:

sudo python3 main.py
INFO     - __main__                       - Initializing MLSC...
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM dsnoop:1
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM dmix:1
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 1}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
INFO     - __main__                       - Initialization finished.
INFO     - __main__                       - MLSC started...
INFO     - libs.audio_process_service     - Found the following audio sources:
INFO     - libs.audio_process_service     - Selected Device: 0 - USB-MIC: Audio (hw:1,0) - 192000.0 Hz
INFO     - libs.device                    - Starting device: Default Device
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
ERROR    - libs.audio_process_service     - Could not init AudioService.
ERROR    - libs.audio_process_service     - Unexpected error in init_audio_service: [Errno -9997] Invalid sample rate
Traceback (most recent call last):
  File "/share/music_led_strip_control/server/libs/audio_process_service.py", line 135, in init_audio_service
    stream_callback=callback
  File "/usr/local/lib/python3.7/dist-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9997] Invalid sample rate
INFO     - libs.output_service            - Starting Output service... Device: Default Device
INFO     - libs.effect_service            - Starting Effect Service component from device: Default Device
INFO     - libs.effect_service            - Effects component started. Device: Default Device
INFO     - waitress                       - Serving on http://0.0.0.0:8080
^CINFO     - __main__                       - Stopping MLSC...
INFO     - __main__                       - MLSC stopped
INFO     - libs.effect_service            - Effects component stopped. Device: Default Device

i think its some sort of conflict with the sample rate but I haven't yet been able to sort that one out...

TobKra96, 2021-11-23:
Please Reopen if the issue is still present.

closed 2021-11-23 by TobKra96