TobKra96/music_led_strip_control #132: Possible script to add devices

bsodergren, 2021-07-16:
A modification to allow a shell script to add a new connected node device when one connects to the private WiFi.

TobKra96, 2021-07-17:
You can achieve such an "automatic connection" by scanning the network for new devices and add it to mlsc:

- Via Rest API call, or
- Stop mlsc manipulate the config.json and start mlsc again.

At the moment I do not want to implement such a feature inside the default mlsc program. I don't want to scan the network of our users for new devices.

bsodergren, 2021-08-09:
the stopping the MLSC writing a new config.json is what im doing, but i was wondering if it's possible to get a more streamline, command line interface to add/remove devices.

But i guess that gives me an excuse to learn python

closed 2021-11-23 by TobKra96