TobKra96/music_led_strip_control #133: Run on windows?

Theagainmen, 2021-07-31:
Hello,

First of all thank you for this amazing open-source project!

I was wondering if it's possible to also install and run the server on a windows 10 device, and just run an ESP8266 client with a LED strip. Currently it doesn't seem possible, or I'm just unable to succesfully install it on windows. Is it possible to adjust the code so it works on windows?
If it's too much work or not possible, is it then possible to use an extern microphone from my windows computer?

benedikt966, 2021-08-08:
I just want to add, that I think this idea would be amazing. So you can use with the right settings the output of your windows pc as an input for your LED Music Analyzer. So you can setup your laptop for a party and use it with spotify and have a better sound quality for the programm. But I didnt built it myself yet, so maybe it still works pretty good with a normal microphone, if you set it next to the box with music.

carlyd95, 2021-08-08:
@benedikt966 you can already set it up to receive audio from other methods.

benedikt966, 2021-08-09:
oh wow, thats cool! Than I have one more reason to build this project in the future. Amazing work and good Instructions by the way. I'm really looking forward to do this project =)

carlyd95, 2021-08-09:
@benedikt966 yeah let me know if you need help, I’ve done it with BlueALSA and Librespot

TobKra96, 2021-11-23:
A tutorial for Loopback is planned. Windows can work, but we do not add support.

closed 2021-11-23 by TobKra96