TobKra96/music_led_strip_control #138: Set effect on group of devices via REST API

fherbet, 2021-11-15:
Hi ! 

1st of all, thanks a ton for this wonderful project ! i have now a main ledstrip driven by a pi3 + 2nd one connected to NodeMCU that are on both side of my sofa and both hidden in vertical lamps! vu meter  effect really rocks while listening to music!

i'm now planning to build many more NodeMCU satellites for other rooms !

currently , i'm trying to integrate this into my domotic station (jeedom) via python scripts and requests library to interact with MLSC REST API.

all works fine except that i'm trying to find a way to set same effect on 2 devices at the same time (and not all devices because they're not in same room.)

obv when i use this code, both devices acts in a synchronized way 

```
def seteffect(data):
    r = requests.post(url + '/api/effect/active', json=data)

# off for all devices
elif sys.argv[1] == ('off2') :
    jsondata={'effect': 'effect_off'}
    seteffect(jsondata)

# gradient for all devices
elif sys.argv[1] == ('gradient2') :
    jsondata={'effect': 'effect_gradient'}
    seteffect(jsondata)
```

when i use that one, i get a delay between both 'seteffect'

```
if sys.argv[1] == ('off') :
    jsondata={"device": "device_0","effect": "effect_off"}
    jsondata2={"device": "device_1","effect": "effect_off"}
    seteffect(jsondata)
    seteffect(jsondata2)

elif sys.argv[1] == ('gradient') :
    jsondata={"device": "device_0","effect": "effect_gradient"}
    jsondata2={"device": "device_1","effect": "effect_gradient"}
    seteffect(jsondata)
    seteffect(jsondata2)
```

looks like it's not possible to have a single json object to push effect on several devices like json object we get when we retrieve devices effect > GET /api​/effect​/active > Return active effect

```
{
  "devices": [
    {
      "device": "device_0",
      "effect": "effect_off"
    },
    {
      "device": "device_1",
      "effect": "effect_off"
    }
  ]
}
```

Am I wrong ? or i just missed something ?

thanks very much !

Teraskull, 2021-11-18:
Hi @fherbet,

Thanks for your feedback!
I just pushed a [commit](https://github.com/Teraskull/music_led_strip_control/commit/a3e054468ecb4ae0a9d58fa1582cabacf75e7af4) to allow setting effects for multiple devices via the API.
Please clone my fork https://github.com/Teraskull/music_led_strip_control and test if your strips won't have the delay.

The body for the API call should be like in this example:
```json
{
  "devices": [
    {
      "device": "device_0",
      "effect": "effect_single"
    },
    {
      "device": "device_1",
      "effect": "effect_gradient"
    }
  ]
}
```

I will also update the API documentation with more examples later.

TobKra96, 2021-11-18:
Merged into branch dev_2.3. Thank you 👍

fherbet, 2021-11-18:
thanks guys ! 

i have tried updated MLSC using below command

`curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b dev_2.3`

but i'm still running stable 2.2 (master)

what would you suggest to run MLSC branch dev_2.3 ?

thanks !

closed 2021-11-19 by fherbet