TobKra96/music_led_strip_control #147: RGBW identification of strip ?

frazerb, 2021-12-16:
I have issue using SK6812 RGBW strip with Raspberry PI 4.

The test that exists in a few places in the library code (if "SK6812" in self._led_strip:) fails.  It works if I change this to (if "sk6812" in self._led_strip:) and this matches the definition of the strip identifiers in libs/config_template.json

For example in libs/effects/effects_single.py

Is this a bug in the code or a feature of my incompetence ?

Teraskull, 2021-12-20:
Yes, you are right, recently we changed all config values to lowercase, and forgot to change the respective values in the codebase. 

Thanks for the feedback, on it.

Teraskull, 2021-12-20:
The issue is now fixed, please use the `dev_2.3` branch.

closed 2021-12-20 by Teraskull