TobKra96/music_led_strip_control #153: Unable to start since new install - ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security'' (/usr/local/lib/python3.9/dist-packages/werkzeug/security.py)

beardiebeardie, 2022-03-30:
I did a clean install of raspberry pi os (latest version) and decided to install version 2. I am getting the following message.
Thank you for any help.

Traceback (most recent call last):
  File "/share/music_led_strip_control/server/main.py", line 15, in <module>
    from libs.webserver.webserver import Webserver
  File "/share/music_led_strip_control/server/libs/webserver/webserver.py", line 1, in <module>
    from libs.webserver.blueprints.authentication_api import authentication_api
  File "/share/music_led_strip_control/server/libs/webserver/blueprints/authentication_api.py", line 1, in <module>
    from libs.webserver.executer import Executer
  File "/share/music_led_strip_control/server/libs/webserver/executer.py", line 1, in <module>
    from libs.webserver.blueprints.authentication_executer import AuthenticationExecuter
  File "/share/music_led_strip_control/server/libs/webserver/blueprints/authentication_executer.py", line 4, in <module>
    from flask_login import LoginManager, UserMixin, login_user, logout_user
  File "/usr/local/lib/python3.9/dist-packages/flask_login/__init__.py", line 16, in <module>
    from .login_manager import LoginManager
  File "/usr/local/lib/python3.9/dist-packages/flask_login/login_manager.py", line 24, in <module>
    from .utils import (login_url as make_login_url, _create_identifier,
  File "/usr/local/lib/python3.9/dist-packages/flask_login/utils.py", line 13, in <module>
    from werkzeug.security import safe_str_cmp
ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security'' (/usr/local/lib/python3.9/dist-packages/werkzeug/security.py)

beardiebeardie, 2022-04-01:
The error was from the new version of Wekzeug 2.1.0 which removed some code needed for this app to run. 
I ended up fixing it by downgrading flask and Werkzeug from 2.1.0 to 2.0.3

pip uninstall flask and pip uninstall Werkzeug 
then pip install flask==2.0.3 - which installed the new version of Werkzeug 2.1.0 again.
pip uninstall Werkzeug 
then pip install Werkzeug==2.0.3

Burraby, 2022-04-02:
I got this error too when I reinstalled MLSC

lelonet111, 2022-04-02:
thank you i also have this problem and i can fix it with this
----------------------------------------------------

sudo pip3 uninstall flask
sudo pip3 uninstall Werkzeug (Probably don't need this)
sudo pip3 install flask==2.0.3
sudo pip3 uninstall Werkzeug
sudo pip3 install Werkzeug==2.0.3
-----------------------------------------------------
how can i fix now this???
when i open a terminal and paste folowing step by step then everything works fine!
-----------------------------------------------------
1.\ cd /share/music_led_strip_control/server

2.\ sudo systemctl stop mlsc

3.\ sudo python3 main.py
------------------------------------------------------
then it works fine with musicreactiv. And also in the general settings i can now pick the audio device. but after i close the terminal it wont work. but still work with non music reactive.  i am a newbie and can not fix that. but some code that run this steps one after other with 10 sec. delay after boot will fix that!?!? what do you think??? 

thank you

Burraby, 2022-04-02:
That's weird, the music effects work for me after I close the terminal... Did you restart after you set the audio input device?

lelonet111, 2022-04-02:
> That's weird, the music effects work for me after I close the terminal... Did you restart after you set the audio input device?

yes, when the pi reboot the mlsc runs becaus when i go to cd /share/music_led_strip_control/server, and run sudo python3 main.py, i recive the eror that mlsc allready runs and that i can stop it with sudo systemctl stop mlsc, just like it should work. but if i paste sudo systemctl stop mlsc and then run sudo python3 main.py it works. i can pick the right soundcard and the neopix runs perfect, when i close the terminal the neopix stops but the non music effects work fine.

thank you for you answer

Burraby, 2022-04-02:
I think the python program will stop running when you close the terminal... Why do you need to stop and start MLSC anyway?

lelonet111, 2022-04-02:
> I think the python program will stop running when you close the terminal... Why do you need to stop and start MLSC anyway?

because i cant pick any soundcard, the sound reactiv mode does not work without select first audio device. with this steps it works.

lelonet111, 2022-04-03:
> I think the python program will stop running when you close the terminal... Why do you need to stop and start MLSC anyway?

so for now i found a way, after reboot go over remote desktop open the terminal, then paste
cd /share/music_led_strip_control/server
----------------------------
sudo systemctl stop mlsc
----------------------------
sudo python3 main.py
----------------------------

when i close the remote desktop it still run, but that must be fixed ;-),

i also try to make a .sh on desktop and let that run auto after boot but that dont work verry well, i can see the the audio sources and pick one but the mlsc service dont work than. 

but i will still try

Teraskull, 2022-04-04:
@lelonet111 What? Of course the script will stop working if you close the terminal, That's why systemd is used.

Just run the service again instead of directly with Python:
```bash
sudo systemctl start mlsc
```

Teraskull, 2022-04-27:
Closing issue since [Flask_Login](https://github.com/maxcountryman/flask-login/) has been recently updated to remove the deprecated `safe_str_cmp`. See [changelog](https://github.com/maxcountryman/flask-login/blob/0f94639858946ea8e921743db684b6da97d46385/CHANGES.md#version-060).
Now it is safe to update the following modules:
```shell
pip install -U Flask_Login==0.6.0 flask==2.1.1 werkzeug==2.1.1
```

closed 2022-04-27 by Teraskull

lelonet111, 2022-05-05:
so i fix it! 

go to crontab -e

in the last row i write 

@reboot sleep 10 && sudo systemctl stop mlsc && sleep 5 && sudo systemctl start mlsc

lelonet111, 2022-05-05:
https://www.youtube.com/watch?v=0LkKdmZzovE

Teraskull, 2022-05-05:
@lelonet111 That's a very bad idea and you shouldn't do it. I already said it's safe to update the modules. Why are you looking for workarounds?

Systemd is smart enough on its own to boot properly with delays after internet connection is established.

How is sleeping going to help an `ImportError` anyways?

lelonet111, 2022-05-05:
@Teraskull Teraskull I also tried the following, but it didn't solve my problem

pip install -U Flask_Login==0.6.0 flask==2.1.1 werkzeug==2.1.1

Teraskull, 2022-05-05:
@lelonet111 Did you try running with `sudo`? Pip installs modules differently for a user and root.