TobKra96/music_led_strip_control #154: [newbie question] How come this MLSC API code doesn't work?

Burraby, 2022-04-03:
```py
import requests

query = {
    'device': 'device_0',
    'effect': "effect_single"
}
requests.post('http://192.168.1.90/api/effect/active', params = query)
```
There is no error, it just doesn't work...

Teraskull, 2022-04-04:
Hi @Burraby,

First, it's because you did not specify the default port in the URL (`8080`, or a different one if you changed it).
Second, you have to pass the query as `json`, not `params`.
The following code should work:
```python
import requests

query = {
    "device": "device_0",
    "effect": "effect_single"
}
r = requests.post('http://192.168.1.90:8080/api/effect/active', json=query)
print(r)
```
This should print `<Response [200]>`.

Burraby, 2022-04-04:
> Hi @Burraby,
>
> First, it's because you did not specify the default port in the URL (`8080`, or a different one if you changed it). Second, you have to pass the query as `json`, not `params`. The [...]

Thank you for the quick reply, this seems to work!

closed 2022-04-04 by Burraby